bambulab/BambuStudio #4643: unable to launch Bambu Studio on my Mac

sansari1959, 2024-08-16:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

McOS 14.6.1

### Additional system information

When I launch the application no window is displayed. Even when I open a project no window is displayed. This happened after I upgraded to the latest MacOS Sonoma 14.6.1 version two days ago.

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1- Launch Bambu Lab
2 - splash screen appears briefly
3- Application menu bar appears
4- No main window is displayed.

### Actual results

The application is running, but the main window does not appear. Even when I try to open a file (which I can select through the file browser), after that nothing happens!

### Expected results

main window should appear

### Project file & Debug log uploads

[log.log](https://github.com/user-attachments/files/16641921/log.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

v2nda, 2024-08-19:
Hello, 

I just updated to MacOS 14.6.1 and I'm having the same issue. Did you manage to find a solution?

lanewei120, 2024-08-20:
@sansari1959 @v2nda 
could you also try prusaslicer to check whether it can launch or not?

and could you delete the config file(/Users/user.name/Library/Application Support/BambuStudio/BambuStudio.conf) and then try again?

v2nda, 2024-08-20:
I was able to launch the app by:
1- clicking on the Bambu Studio icon on the dock 
2- clicking view on the top left corner 
3- Enter fullscreen
 This is not a permanent solution but I'm quite sure Bambu will release a new update that would fix this issue.

sansari1959, 2024-08-20:
That indeed works for me. Thanks a lot! This is very helpful.