bambulab/BambuStudio #4644: Tree support not obeying top Z distance, causing poor overhang performance

FJJWFP, 2024-08-17:
### Bambu Studio Version

1.09.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

Bambu Lab A1, Bambu Lab A1 mini

### How to reproduce

On fresh install of Bambu Studio using default configurations, import a part and enable support, chance support style to tree support. Change support top Z distance. Independent support layer height is enabled by default. 
[image][image]
[image]
ght is enabled by default. 

From the above images it can be seen that with a layer height of 0.3, the normal support has a top Z distance of 0.2mm as per the setting, while the tree support rounded the distance up to 0.3mm, which is incorrect and results in poor overhangs. 

### Actual results

The distance between the support interface material and the part is always rounded up or down to the nearest part layer thickness, e.g. a part layer height of 0.3mm will cause all support top Z distance between 0.16 to 0.35 to be rounded back to 0.3mm. 

This issue is present with all types of tree support from my testing, while normal support is not affected. I've also found that the independent support layer height appears to not work with tree supports, as no matter what settings, the support layer height is in sync with the part. Normal support varies layer height as expected. 

### Expected results

The distance between the support interface material and the part should change according to the setting, irrespective of the part layer height. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16644230/log.zip)
[support test.zip](https://github.com/user-attachments/files/16644231/support.test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

FJJWFP, 2024-08-26:
This behaviour is still present with the latest update (1.9.4.60) and does not appear to have changed

tangketan, 2024-08-29:
Yes, accurate top z distance requires variable support layer height, which is absent in organic support. Please use the other styles for now.