bambulab/BambuStudio #4646: Import changes active file resulting in loss of user data

kotarak, 2024-08-18:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Linux bjalo 6.10.0-artix1-2 #1 SMP PREEMPT_DYNAMIC Tue, 23 Jul 2024 20:20:14 +0000 x86_64 GNU/Linux

### Additional system information

CPU: AMD Ryzen 7 4700U
Memory: 8GB

### Printer

Bambulab A1 mini

### How to reproduce

Note: the contents of the named files are irrelevant. Their source was in my case Ondsel.

1. Open Bambulab with empty project.
2. Save empty project as please_save_here.3mf.
3. Import X_X.3mf.
4. Import 2nd_import.3mf.
5. Save project.

### Actual results

X_X is overwritten causing unexpected loss of user data.

### Expected results

Project is saved in please_save_here since no "Open Project" action was done. Only import.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16647339/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

MackBambu, 2024-08-21:
Do you mean that when importing the 2nd_import.3mf file, it doesn't pop up this prompt window below?
[image]

kotarak, 2024-08-22:
The above is not shown during the first import. However, it is on the second. I created [a small example](https://github.com/user-attachments/files/16708326/example.zip).

Steps to reproduce:

1. Unzip the attached example.
2. Open Bambustudio with a new empty project.
3. Save the empty project as project_expected_here.3mf.
4. Import 1st_import.3mf. (Note: Above window *is not* shown.)
5. Import 2nd_import.3mf. (Note: Above window *is* shown.)
6. Save the project.

Expected result: The saved project is found in project_expected_here.3mf.
Actual result: 1st_import.3mf is overwritten without further warning. project_expected_here.3mf remains empty.

I hope this is now more clear.

MackBambu, 2024-08-22:
Since project_expected_here.3mf is an empty project, when importing the 1st_import.3mf file, it can only select to be imported as a project and will not show the selection window.
The same logic applies to the upstream software Prusa.

kotarak, 2024-08-22:
"Open" and "Import" are two very different operations and the latter acting like the former without any indication to the user is a problem. That the selection window is not shown (Why not? Maybe I want only the geometry to be imported?) is separate aspect. My main issue is that this unexpected and unintiutive behaviour caused the loss of data for the user without any warning or other 
feedback from BambuStudio

And the actions of the user are quite reasonable, aren't they? Save a new project. Import stuff. Save the project again. What in this sequence makes you believe that the active project is suddenly something different?

Luckily I could easily reconstruct the overwritten file. But what if that is not possible and the user doesn't have a backup? Super-GAU.

If Prusa is the base for BambuStudio, you might want to let them know of the issue, so that they can fix it and the change can trickle down at some point. Or you decide it's not an issue or too many people really on that behaviour, then you don't tell them.

MackBambu, 2024-08-23:
Prusa has identical discussion threads that can be merged together and discussed as an improvement.