bambulab/BambuStudio #4649: Allow for 4 filaments during Automated Flow Rate Calibration

JPotter73, 2024-08-18:
**Issue:** (using X1) Currently I can only select a single filament of 4 to do Flow Rate Calibration using the Automated method. This is time consuming since my printer is upstairs and my computer is down stairs

**Proposed Solution:** Use the full bed to allow for up to 4 filaments (same AMS) to be calibrated at the same time

**Describe alternatives you've considered**
None comes to mind.

**Additional context**
the Flow Dynamics allows up to 4 filaments to be calibrated at a time with checkboxes to select/deselect the filaments to calibrate. This is fantastic and I wish the same was available for Flow Rate auto. The manual Flow Rate would have to be limited to 1 at a time since it requires more bed space.

Qjeezy, 2024-08-19:
I agree with this! It’s quite annoying that it cannot do all 4 like the dynamics calibration can. There is plenty of build plate space in front of the light to do 4 tests.

Doyle4, 2024-08-21:
Curious, is it possible to manually calibrate each filament, when the AMS changes to a different filament it uses the profile?
If so, be good to see an AMS Flow Rate Calibration, each filament is Auto checked,  each result is then saved to the current filament profile and used when using an AMS.