bambulab/BambuStudio #4650: Auto calibration of Flow Dynamics doesn't flush between filaments

JPotter73, 2024-08-18:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

NA - Not important to issue

### Printer

Bambu X1 (Kickstarter)

### How to reproduce

Choose automatic dynamic flow calibration and more than 1 filament. In this case, AMS slots 3&4 on 2nd AMS (but I've seen this before when I had 1 AMS)

### Actual results

The filament is cut and changed, but not flushed. I was sure to add both filaments in Studio and set a flushing volume, but this still  happens (2nd colour calibration contains 1st colour)
[image]

### Expected results

Clean colours on all test patterns. If the colour isn't fully transitioned, it could be due to flushing volumes, but I had that defined on my run using the Autocalc and 80% (always seems to work for me)

### Project file & Debug log uploads

This does not require a project file or log file. It is inherent in the embedded filament calibration tools.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Qjeezy, 2024-08-19:
This definitely needs to be fixed. For me, it only happens on the second filament calibration. It doesn’t flush enough and half of the calibration is still filament from the first calibration. The third and 4th calibrations do not have this issue, just the second one.

JPotter73, 2024-08-20:
@Qjeezy : good to know! it does mean I can calibrate states 1, 3 and 4 at the same time but do 2 separately. Still a bug to address for sure!

Qjeezy, 2024-08-20:
> @Qjeezy : good to know! it does mean I can calibrate states 1, 3 and 4 at the same time but do 2 separately. Still a bug to address for sure!

I'm not quite sure if that would work. There's still going to be a second test print.

JPotter73, 2024-08-24:
@Qjeezy : I'm quite sure that if I choose slots 1,3,4 only, it skips the 2nd position. However, does that mean it properly flushes for 3? or does the same problem happen? I can't remember. I'll have to keep an eye on it to see. I usually only run 1 when loading a new spool, but sometimes I load new sets for a specific project, so it's def good to solve this one, but I can certainly work around it doing 1 at a time. Just have to go up and down the stairs from printer to PC :)

zhimin-zeng-bambulab, 2024-09-06:
Each time you change the material, you have purged the necessary amount of purge, and the remaining part, the prime line will purge again, and it will be clean when you really calibrate, even if there is still some residue, the impact on the overall is very small.
So it will not affect the final calibration result.

zhimin-zeng-bambulab, 2024-09-06:
the same as https://github.com/bambulab/BambuStudio/issues/2148

JPotter73, 2024-09-07:
@zhimin-zeng-bambulab : if this changed in the last 3 weeks, then great. However, the image shows that the brown is fully not flushed when starting the white and it carries through for almost the entire calibration pattern. It only does this on the 2nd position in calibration, from what i recall.

be221b, 2024-09-21:
Still a bug in 1.9.7.52

Qjeezy, 2024-09-22:
> Still a bug in 1.9.7.52

They basically told us it's not an issue. It definitely is if the filaments are different enough imo. It flushes enough between the other tests, why can't they adjust it to flush enough between the first and second one.

be221b, 2024-09-22:
Seems like it should be an easy bug to fix. It definitely wasn't always like this. At some point, you could calibrate all 4 filaments in an AMS without this issue.

be221b, 2024-09-22:
> > Still a bug in 1.9.7.52
> 
> They basically told us it's not an issue. It definitely is if the filaments are different enough imo. It flushes enough between the other tests, why can't they adjust it to flush enough between the first and second one.

Seems like it should be an easy bug to fix. It definitely wasn't always like this. At some point, you could calibrate all 4 filaments in an AMS without this issue.

Qjeezy, 2024-09-22:
> Seems like it should be an easy bug to fix. It definitely wasn't always like this. At some point, you could calibrate all 4 filaments in an AMS without this issue.

Definitely seems like an easy fix. All they gotta do is adjust the flush amount in the calibration code. I wish they'd give us access to the calibration code, I'd do it myself lol. I do remember a while back that it did work normal like you said.

gaolegao-lx, 2024-09-30:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

be221b, 2024-10-18:
This does not require a project file or log file. It is inherent in the embedded filament calibration tools.