bambulab/BambuStudio #4652: Sehr sehr lange Druckzeit bei Teilen, welche nur eine Linienbreite dick sind.

Pady333, 2024-08-19:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C/P1S

### How to reproduce

1. Erstelle einen Ring, welcher einen Innendurchmesser von 30 mm hat und eine Wandstärke von 0,4mm dick ist.
2. Exportiere dies aus dem CAD Programm als Objektdatei (.obj)
3. Öffne Bambu Studio
4. Ziehe die Objektdatei in das Bambu Studio hinein und erstelle 2 Kopien davon
5. Ändere die Druckreihenfolge auf "nach Objekt"
6. Klicke jetzt auf "automatisch Anordnen"
7. Klicke auf "Druckplatte slicen"

### Actual results

Sehr sehr lange Druckzeiten, die nicht stimmen werden

### Expected results

Normale Druckzeiten.

### Project file & Debug log uploads

[Parametric TPU-Ring sehr lange Druckzeit.zip](https://github.com/user-attachments/files/16662071/Parametric.TPU-Ring.sehr.lange.Druckzeit.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-08-20:
[image]
[image]
The printing time is about 10 minutes, including about 5 minutes of preparation time and about 5 minutes of printing time。
Die druckzeit beträgt etwa 10 minuten, einschließlich 5 minuten vorbereitungszeit und 5 minuten druckzeit。

Pady333, 2024-08-20:
Ich habe es heute nochmals probiert und jetzt funktioniert es wieder. Wahrscheinlich kommt es von einem annähernden vollen Arbeitsspeicher oder ähnlichem

closed 2024-08-20 by Pady333