bambulab/BambuStudio #4653: Add support for 1.0mm, 1.2mm, and 1.4mm nozzle sizes.

Qjeezy, 2024-08-19:
A lot of us have started using the panda Revo hot ends and this allows us to use larger nozzles, 1mm+. The Slicer and Printer does not allow us to select anything larger than 0.8mm. 

**Describe the solution you'd like**
I would like to see support and pre built printer & print profiles for nozzle sizes 1mm, 1.2mm, and 1.4mm. 

**Describe alternatives you've considered**
None. 

**Additional context**
It would be nice to be able to swap in my 1.0mm nozzle, pick the printer profile for it, and print just like we already do for the other smaller size nozzles.

julianito83, 2024-08-26:
Yes! I am dying to use my 1.4 nozzle!!!

ItsTwisted1, 2024-11-01:
I would love this option to

lukigold, 2024-12-08:
Also waiting for this would be awesome

Cenwulf, 2025-01-08:
Would love to see support for 1.5 and 2.5 mm nozzles too. Honestly just allowing users to add custom nozzles for bambu lab printers using the “create printer” dialogue would be great.