bambulab/BambuStudio #4654: Speeds for overhang walls not used

Apulo, 2024-08-19:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10/11

### Printer

Bambu Lab X1 Series/P1P

### How to reproduce

These are the current speed settings I used for the test. I set bridge speed to an insanely high value to highlight the issue.
[image]

As you can see, after slicing the outer walls of the model are correctly identified as overhang walls.
[image]

However, the slicer is printing those walls at the bridge speed, not the defined overhang speeds.
[image]

### Actual results

Overhang walls are printed as bridges, not overhangs

### Expected results

When a wall is identified as an overhang wall, they should be printed using the "Slow Down for Overhangs" settings, not printed as a bridge.

### Project file & Debug log uploads

[scandic_plate_overhang_test.zip](https://github.com/user-attachments/files/16664561/scandic_plate_overhang_test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-08-21:
@Apulo exactly, I am improving it.

Apulo, 2024-08-21:
Great to hear! Thanks for responding so quickly.

tietz-r, 2024-08-24:
Hi,
i have the same issue.
I noticed that this bug appears if you enable support, even if no support will be generated.
If support is disabled, it behaves as intended.
I hope this will help

QingZhangBambu, 2024-08-27:
@Apulo @tietz-r both bugs i have fixed. you can download the latest version to check

QingZhangBambu, 2024-08-27:
[image]
[image]

closed 2024-08-30 by QingZhangBambu