bambulab/BambuStudio #4655: Missing tree supports

alejandrojrey, 2024-08-19:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Slice atached 3mf

### Actual results

Tree suports: 

[image]

Normal supports:

[image]

### Expected results

Generate tree supports as in normal supports

### Project file & Debug log uploads

[supports.zip](https://github.com/user-attachments/files/16664707/supports.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Pady333, 2024-08-20:
[image]
Wenn der letzte Punkt bei Support nicht angeklickt wird, dann sind die Supports wieder da!

[image]

MartinChan3, 2024-08-26:
@Pady333 The checkbox you clicked means 'remove little overhangs', studio will detect the overhang size and if it is small than a threshold then it will not generate supports under these overhangs.If you want all the small overhangs should be considered, do what you did or use manual support to make sure all the overhangs you want to add support

alejandrojrey, 2024-08-26:
> @Pady333 The checkbox you clicked means 'remove little overhangs', studio will detect the overhang size and if it is small than a threshold then it will not generate supports under these [...]

The same overhang is on the y axis, so it should generate the supports on both axis or none.