bambulab/BambuStudio #4656: Dark mode makes ui effect the same color

mike6mike, 2024-08-19:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1

### How to reproduce

When using dark mode, it causes the selected print profile highlight effect to be the same dark background color.
[image]

### Actual results

Invisible ui effect in dark mode

### Expected results

When not in dark mode, the background is usually white. So when you click on a print profile, it gives you a visual representation/cue that it's correctly selected before downloading.

### Project file & Debug log uploads

[Flipper v2.zip](https://github.com/user-attachments/files/16665391/Flipper.v2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-08-20:
HI [mike6mike](https://github.com/mike6mike)：
[image]
Do you mean  that the background color of this area, which is the same as the dark pattern becomes black, it's not recommended to show it as black

mike6mike, 2024-08-20:
Hi [ttt888zzz](https://github.com/ttt888zzz):

[image]
I'm talking about the area right below the yellow marked box you posted.