bambulab/BambuStudio #4657: Online Models 404 error

KoProGnosis, 2024-08-20:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Use the Bambu Studio Application in a PC with Greek region
2. Go to Home page
3. Click in a model in "Online Model" category

### Actual results

An error 404 occurs. The page cannot be found. It seams like the program tries to open the webpage in Greek, but the is not a Greek version so the problem occurs. 
For example this is a URL that the app creates:
[](https://makerworld.com/en/el/studio/webview?from=bambustudio&modelid=596767)
If we remove the **'/el'** from the URL, like this:
[](https://makerworld.com/en/studio/webview?from=bambustudio&modelid=596767)
the page loads fine. 

### Expected results

The URL generator should not change the URL to unsupported languages.  

### Project file & Debug log uploads

There is not such file for this error. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-08-21:
HI [KoProGnosis](https://github.com/KoProGnosis):
Could you please provide a video ,we will check it

KoProGnosis, 2024-08-21:
[image]
The video was above 10mb so I hope this gif helps.

ttt888zzz, 2024-08-21:
HI [KoProGnosis](https://github.com/KoProGnosis):
the region is Europe，right？

[image]
[image]

KoProGnosis, 2024-08-21:
[image]
The video was above 10mb so I hope the gif will do the job.

KoProGnosis, 2024-08-21:
Hello @ttt888zzz, the language setting was undefined, I changed it to English and now it works.  The Region was already set to Europe. Thank you for your time and the solution!

KoProGnosis, 2024-08-21:
Fixed.

ttt888zzz, 2024-08-21:
I’m happy to help

closed 2024-08-23 by ttt888zzz

Deepcuts, 2024-12-27:
December 2024 and same issue. Language undefined gives 404 everywhere.
Thanks to https://github.com/bambulab/BambuStudio/issues/4657#issuecomment-2301779129 I fixed it, but it owuld be nice if Bambu Studiu got fixed instead.