bambulab/BambuStudio #4659: The color doesn't match my paint

landofbosses, 2024-08-20:
### Bambu Studio Version

01.06.01.00

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11, version 23H2

### Additional system information

[image]
[image]

### Printer

bambu p1s 

### How to reproduce

paint your 3d print and print

### Actual results

look at pics

### Expected results

the painted print in the slicer

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-08-21:
HI [landofbosses](https://github.com/landofbosses)：
Could you provide the source file，we will check it

landofbosses, 2024-08-21:
Thank you for the response. I figured out the problem. For some reason the
filament flush went to zero

closed 2024-08-23 by ttt888zzz