bambulab/BambuStudio #4664: Wall count doubles with modifiers over 3 walls

MatthaeusLeiner, 2024-08-22:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1E

### How to reproduce

Insert a Zylindrical Modifiers and up the wall count above 3. 

[image]

### Actual results

With 3 walls:

[image]

With 4 walls:

[image]

It also creates a gap on the outside wall with 4 walls:

[image]

### Expected results

It should have only the inserted wall amount.

### Project file & Debug log uploads

[Wallerror.zip](https://github.com/user-attachments/files/16707605/Wallerror.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-08-23:
yes, it does a problem. but a bit hard to fix. I will add it to our fix list.

vanowm, 2026-02-04:
@zhimin-zeng-bambulab How is this a feature request? It's clearly a bug