bambulab/BambuStudio #4666: CLI Assembly process does work, fails with Segmentation fault: 11

rizwansarwar, 2024-08-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

OS: MacOS Sonoma 14.5

### Additional system information

Apple MacBook Pro M1 (Apple Silicon)

### Printer

Bambulab P1P

### How to reproduce

1) Could not attach STL's but they are all just 20x20 cubes primitives exported from BS. Export a cube primitive from BS and make 4 copies of it. naming them 1.stl, 2.stl, 3.stl, 4.stl, 5.stl
2) Run the attached command using the attached, generic_pla.json, process_p1p.json, machine_p1p.json and Assembly.json. 
`/Applications/BambuStudio.app/Contents/MacOS/BambuStudio --export-3mf out.3mf --load-settings="machine_p1p.json;process_p1p.json" --load-filaments="generic_pla.json;generic_pla.json;generic_pla.json;generic_pla.json;generic_pla.json" --filament-colour="#000000;#FF0000;#33CC33;#FFFF00;#0000FF" --load-assemble-list="Assembly.json" --debug 5`

[Assembly.json](https://github.com/user-attachments/files/16726765/Assembly.json)
[generic_pla.json](https://github.com/user-attachments/files/16726771/generic_pla.json)
[process_p1p.json](https://github.com/user-attachments/files/16726783/process_p1p.json)
[machine_p1p.json](https://github.com/user-attachments/files/16726788/machine_p1p.json)

3) The process fails with Segmentation Fault 11.

### Actual results

The process fails with Segmentation Fault 11. I have attached the output from slicer.
[output.txt](https://github.com/user-attachments/files/16726815/output.txt)

### Expected results

The process should result in a 3mf file with specified assembly. 

### Project file & Debug log uploads

[output.txt](https://github.com/user-attachments/files/16726853/output.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-08-24:
these config json need to be a full file, not the original one in the profile directory
for example, here is the full list file of generic_pla.json for p1p
[generic_pla_p1p_1.json](https://github.com/user-attachments/files/16735365/generic_pla_p1p_1.json)

rizwansarwar, 2024-08-24:
@lanewei120 How do I obtain these config files for my filament for example? There seems to be no clear guideline as to how to obtain them?

lanewei120, 2024-08-27:
usually my colleague uses some python tool to generate the full config file from those files
and I just asked this file from them

rizwansarwar, 2024-08-27:
@lanewei120 Thank for doing that. It would good if it was available to everyone or a guide on how to obtain these files as the feature is very useful.

lanewei120, 2024-08-27:
I think you can ask ChatGpt to do it using python

rizwansarwar, 2024-08-30:
@lanewei120 I don't really know how to do that. I have tried the export feature of BambuSlicer to export filament presets, that does not really work. The JSON it outputs does not work with CLI. It is a bit confusing and misleading that Wiki refers to these files to be "FULL JSON", but then it is clear how to obtain these files. 

btw, I tried to use your file and it worked, I just don't know how to get the same file for different supported printers for Bambu Studio such as Prusa or creality printers.