bambulab/BambuStudio #4667: Treating Overhangs as Bridges

CS-Account, 2024-08-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

NA

### Printer

X1X

### How to reproduce

https://youtu.be/HcSOz-Lsxgg?si=ewtu1FRl8s-wOHfh&t=134

This exact point of this video demonstrates and explains the problem. This has apparently been fixed already in Prusa slicer, but the fix has not been brought over.

### Actual results

Shown in video

### Expected results

The overhang should be treated as an overhang.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jedthe3rd, 2024-08-24:
I noticed this issue when I was tuning my overhang parameters on my X1C that arrived this week.

MartinChan3, 2024-08-26:
@CS-Account I believe this question is same with  #4654

CS-Account, 2024-08-26:
@MartinChan3 I believe both things are caused by the same underlying bug yes. Both mean that overhangs look worse in BBS vs Prusa slicer which I believe fixed this some time ago, as the section of the video I linked shows. I think the dates in the video could help track down which exact changes fixed the problem.

CS-Account, 2024-08-26:
Duplicate of #4654

closed 2024-08-26 by CS-Account

MartinChan3, 2024-08-26:
@CS-Account The new version has been released and the question seems can be solved with new options, try it my friend!