bambulab/BambuStudio #4669: No 3D view

Buom01, 2024-08-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Built from source by yourself or others

### OS version

CachyOS Linux x86_64

### Additional system information

[image]

Compiled from aur/bambustudio-git

### Printer

Bambu A1 mini

### How to reproduce

1. Open Bambu Studio
2. Go to Prepare or Preview tab

### Actual results

No 3D view at all

### Expected results

Ability to view my model with the 3D view

### Project file & Debug log uploads

[debug_Fri_Aug_23_18_50_35_37379.log](https://github.com/user-attachments/files/16731643/debug_Fri_Aug_23_18_50_35_37379.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Buom01, 2024-08-23:
https://www.reddit.com/r/BambuLab/comments/1djnpvv/comment/ljkmltt/?context=3

Buom01, 2024-08-23:
Tags: Multi-monitor, multiscreen, dual monitors, ultrawide, dual boot, dualboot

Knifa, 2024-08-26:
Also experiencing this after upgrading to `nvidia 560.35` on Arch.

IvanVojtko, 2024-08-26:
Downgrading to nvidia 555 fixed the issue.

Knifa, 2024-08-26:
Same for me, back to 555 fixed it. Multi-monitor, etc. made no difference for me as per the Reddit thread?

lollo03, 2024-08-31:
I can confirm that this also happens on arch with the latest nvidia-open driver when using wayland.
If I use a regular xorg session the 3d view functions properly.

Noodle-Bug, 2024-09-05:
Rolling back to the newest 555 nvidia drivers fixed this for me as well (on Arch). This also isnt the only issue these drivers have caused for me.

https://archive.archlinux.org/packages/n/nvidia-utils/nvidia-utils-555.58.02-1-x86_64.pkg.tar.zst
https://archive.archlinux.org/packages/n/nvidia-dkms/nvidia-dkms-555.58.02-1-x86_64.pkg.tar.zst

For anyone else on Arch + Nvidia GPU having this issue, I just used [downgrade](https://github.com/archlinux-downgrade/downgrade) to roll back from tty and all seems well now:
`sudo downgrade "nvidia-utils=555.58.02-1" "nvidia-dkms=555.58.02-1"`

makkarpov, 2024-09-06:
For anyone wondering how to use BambuStudio and still keep 560 drivers: switch it to software rendering:

```
__GLX_VENDOR_LIBRARY_NAME=mesa  __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
```

This works on Fedora 40 for me.

Buom01, 2024-09-13:
Thanks for all participation.
I'm done with Nvidia. I cannot even have a TTY screen working as it just turn all my display off, and all the latest drivers I tested just doesn't work at all (I don't talk about negligible artifact). Furthermore, I tested a lot of things from forums, and no one solution worked for me.

With official Nvidia driver, on Plasma X11 I don't even have decoration showing. With Wayland, Plasma just doesn't start.
I don't talk about other drivers and desktop env because I lost three days of my life on it and doesn't work in any way.
I don't want to mess up with old driver version, but I have only tested quickly 555 version.

TL;DR
# NVIDIA ⇒ 🗑️
Hello AMD

Closing as it's seem to be a pure Nvidia issue, and thus nothing to do with BambuStudio.

makkarpov, 2024-09-13:
Definitely not a BambuLab issue, considering that BambuLab is the only application that is having troubles with nvidia drivers on my system. Plasma works, FreeCAD works, Cura works, Slic3r works, PrusaSlicer works, BambuLab dont.

Considering the overall quality level of BambuLab software, I'm not surprised to see that quality level of the management is on par with it.

RamboRogers, 2024-09-23:
FIX FOR KDE / Arch

Paste this, into the ENV in the shortcut.
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1`

1. [image]
2. [image]
3. Run it
4. It works
[image]

codywlsmith, 2024-09-23:
> **FIX FOR KDE / Arch**
>
> Paste this, into the ENV in the shortcut. `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 [...]

Neither adding these variables to the Environment Variables in the shortcut, or adding them individually in Flatseal fixes the issue for me.

jameskitt616, 2024-09-23:
I also wonder how to get it to work with flatpak/flatseal.
Appreciate any help.

JaimeHW, 2024-10-13:
Same. Makes the software entirely unusable for us.

MackBambu, 2024-10-31:
I installed the 560 driver on Ubuntu 24 and ran it in Wayland mode. However, the issue did not reoccur, so it seems there is additional information that has not yet been identified.

different-name, 2024-11-10:
@Buom01 would you consider reopening this issue as others are still experiencing this Bambu Studio + Nvidia specific issue? Unfortunately I'm not in a position to order a new GPU :smiling_face_with_tear:

yodatak, 2024-11-20:
same for me on bazzite fedora 41 and nvidia proprietary driver Driver Version: 565.57.01 and wayland

jl1990, 2024-11-22:
Any update on the progress of the fix for this issue? Downgrading nvidia driver is not an option for everybody.

Thanks!

MackBambu, 2024-11-24:
> Any update on the progress of the fix for this issue? Downgrading nvidia driver is not an option for everybody.
> 
> Thanks!

Updating the OpenGL version is in progress and requires a lot of work.

Hamid-0, 2024-11-27:
> For anyone wondering how to use BambuStudio and still keep 560 drivers: switch it to software rendering:
> 
> ```
> __GLX_VENDOR_LIBRARY_NAME=mesa  __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
> ```
> 
> This works on Fedora 40 for me.

This also works for Orca slicer V2.2.0

Seann-Moser, 2024-12-02:
> **FIX FOR KDE / Arch**
>
> Paste this, into the ENV in the shortcut. `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 [...]

I was able to get this on Hyperland/Arch using an alias 
```
bambu='__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 bambu-studio'
```

tyumici, 2024-12-05:
Chiming in, this issue is present in Arch on the current AUR. I'm using KDE and running a X11 session instead of Wayland works but is annoying to need to do. The above fixes with Env vars did not work for me.

different-name, 2024-12-06:
> The above fixes with Env vars did not work for me.

@tyumici does the `/usr/share/glvnd/egl_vendor.d/50_mesa.json` file exist on your system?

MasonMayjack, 2024-12-18:
> **FIX FOR KDE / Arch**
>
> Paste this, into the ENV in the shortcut. `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_DMABUF_RENDERER=1 [...]

That has stopped it starting entirely for me

Jannomag, 2024-12-25:
Works for me on EndeavourOS with KDE Plasma 6.2.4 on Wayland. But when going into the preview mode it gets very laggy and barely usable.
I have an AMD 3800X, Nvidia RTX 3070, 32GB RAM. So the software renderer seams to have very low performance. Using it on my laptop with an i5-7200U without dedicated graphics and the same EOS version works way better. I also tried the ENV line on this device.

Any ideas how to improve it?

MasonMayjack, 2024-12-26:
Yeah, nah. I'm just a moron and was deleting the env line that points to
the executable. It works now and I'm not lagging too bad with a 3050 and an
ryzen 5600x

js01, 2025-01-01:
I have an ASUS TUF F15 laptop with an i5-12500H, a GeForce RTX 3050, and 32GB RAM. 
NVIDIA Driver v565.77. 
EndeavourOS 6.12.7-arch1-1 (64-bit), KDE Plasma 6, Wayland

The system is set to use the discreet GPU only. I installed Bambu Studio via AUR and none of the fixes in the thread worked for that version. Setting the environment variable posted did not work.
I then removed it and installed the flatpak version and using Flatseal, changed the environment variable there, and the 3D viewports all work now.

`__GLX_VENDOR_LIBRARY_NAME=mesa  __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
`

jameskitt616, 2025-01-02:
> I have an ASUS TUF F15 laptop with an i5-12500H, a GeForce RTX 3050, and 32GB RAM. NVIDIA Driver v565.77. EndeavourOS 6.12.7-arch1-1 (64-bit), KDE Plasma 6, Wayland
> 
> The system is set to use the discreet GPU only. I installed Bambu Studio via AUR and none of the fixes in the thread worked for that version. Setting the environment variable posted did not work. I then removed it and installed the flatpak version and using Flatseal, changed the environment variable there, and the 3D viewports all work now, though it is laggy. Interface works fine otherwise.
> 
> `__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json `

Is all you changed adding those two env variables via flatpak? because when i do it, it still doesnt show and 3d viewport

js01, 2025-01-03:
> Is all you changed adding those two env variables via flatpak? because when i do it, it still doesnt show and 3d viewport

Actually, after rechecking, the only thing I added in Flatseal was this line in Bambu Studio > Enviroment:
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json`

I don't know how or why I omitted the `__GLX_VENDOR_LIBRARY_NAME=mesa` part, but it works with just the line I've posted. I also enabled `GPU Acceleration` under Bambu Studio > Device.

jameskitt616, 2025-01-03:
For me the `/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json` file does not exist, only the file you mentioned previously `/usr/share/glvnd/egl_vendor.d/50_mesa.json`
For me this file contains the following:
```
{
    "file_format_version" : "1.0.0",
    "ICD" : {
	"library_path" : "libEGL_mesa.so.0"
    }
}
```
does the other file of yours also contains the same infos?

js01, 2025-01-04:
> For me the `/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json` file does not exist, only the file you mentioned previously `/usr/share/glvnd/egl_vendor.d/50_mesa.json` For [...]

The entire directory `/usr/lib/x86_64-linux-gnu/` does not exist on my system.
The files and directory for `/usr/share/glvnd/egl_vendor.d/50_mesa.json` do exist and `50.mesa.json` is identical to yours.

I had switched to `HYBRID GPU mode`, so I think Bambu was using that and it was working fine.
I have switched back to `dGPU (ASUS MUX)` and here is what I've found:

Having `__GLX_VENDOR_LIBRARY_NAME=mesa` makes no difference when using either line, so I've omitted it. I've tried combinations of the three and having it made no difference.

`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json`
It just works, despite that file not existing. I thought I had seen somewhere in this thread or another discussion that the fix may be just to point it to a non-existant or empty file, and that seems to work for me. HOWEVER, if I change `50_mesa.json` to ` blank.json` for example, Bambu stops working. It only works as it is written, despite the file being nonexistent. I don't get it.

Using `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json` at any point breaks Bambu completely. I get no front page, no viewports, and it doesn't connect to my printer.

jameskitt616, 2025-01-04:
Thats really interessting, i've also set `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json` now in flatseal to bambulab, eventho the file/path also doesnt exist for me, and it works now as well.

raviboth, 2025-01-05:
Same for me, also reporting this issue on bazzite with latest nvidia drivers and `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json` in flatseal environment variable temporarily fixes it

BradleyUffner, 2025-01-06:
I had to use `/usr/share/glvnd/egl_vendor.d/50_mesa.json` as the path for `__EGL_VENDOR_LIBRARY_FILENAMES` instead of the one specified above to get it working on Fedora 41.

Leonw43, 2025-01-08:
Same issue on Fedora 40 & 41 - the workaround is software rendering which degrades performance quite significantly.

Stoiker, 2025-01-31:
Same issue here on Fedora 41 + Wayland with GeForce 1070.
Any updates on that?

Leonw43, 2025-01-31:
> Same issue here on Fedora 41 + Wayland with GeForce 1070. Any updates on that?

I ended up just using CachyOS/Arch since it lets you pick an X11 session instead of forcing wayland. They need to fix this as X11 is on borrowed time..

fdev31, 2025-01-31:
Same issue here  on Arch/Hyprland/RTX 4080/Nvidia 565 driver

Stoiker, 2025-01-31:
Does anyone have more technical insight into what might be the problem here? This sound familiar to something occurred on FreeCad (https://forum.freecad.org/viewtopic.php?style=10&t=89408)

Thus more information would be great (beside the common knowledge that NVIDIA and Wayland don't get along very much)

CMasterson, 2025-02-16:
> Actually, after rechecking, the only thing I added in Flatseal was this line in Bambu Studio > Enviroment: `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/default/share/glvnd/egl_vendor.d/50_mesa.json`

This worked for me.
Bazzite 41
Nvidia 4090 (570-86-16)

All other settings are stock on Flatseal.

[image]

ajunca, 2025-02-21:
I don't know why github is not referencing my other issue comment #4626. But I found a solution that works for me, under wayland + nvidia that is hardware accelerated. I found it looking in orca slicer issues, and it is using zink layer, opengl on top of vulkan instate of nvidia opengl implementation (not sure if all env variables are needed, but as it is, is working for me):

```
__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink WEBKIT_DISABLE_DMABUF_RENDERER=1 bambu-studio
```

whimsicalwyrm, 2025-02-25:
@ajunca Thanks, this works for both Bambu Studio 01.10.01.50 and Orca Slicer 2.3.0-Beta ! I have finally hardware acc so it doesn't lag. Tested a bit and only 3 settings are needed:

__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json MESA_LOADER_DRIVER_OVERRIDE=zink WEBKIT_DISABLE_DMABUF_RENDERER=1

First gets 3d preview working, second enables hardware acc and third is needed to see home page contents.

animusfracto, 2025-02-25:
OS is bazzite-nvidia:41 (an Atomic Fedora variant) and an nVidia card, running Bambu Studio via flatpak.  I haven't been able to get hardware acceleration to work with any of the environment variables mentioned here, guess I'm gonna have to live with software rendering for the time being.

Leonw43, 2025-02-26:
@ajunca Thanks - confirming this worked for me also, on CachyOS, RTX3070 and 570.86

hydrohs, 2025-03-14:
> I don't know why github is not referencing my other issue comment [#4626](https://github.com/bambulab/BambuStudio/issues/4626). But I found a solution that works for me, under wayland + nvidia that [...]

This worked for me as well, much appreciated.

IvanVojtko, 2025-04-01:
For arch linux users, this package has correct env variables to run fine using nvidia cards
`https://aur.archlinux.org/packages/bambustudio-nvidia-bin`

dmcgoldrd, 2025-05-23:
> For arch linux users, this package has correct env variables to run fine using nvidia cards `https://aur.archlinux.org/packages/bambustudio-nvidia-bin`

This was working but now is giving LibOSmesa dependency issues, any ideas?

tyumici, 2025-05-24:
The regular version appears to also have this issue

dmcgoldrd, 2025-05-25:
> The regular version appears to also have this issue

It's caused by the upgrade to LibMesaOS. A downgrade in the manager resolved the issue

IvanVojtko, 2025-06-02:
> > For arch linux users, this package has correct env variables to run fine using nvidia cards `https://aur.archlinux.org/packages/bambustudio-nvidia-bin`
> 
> This was working but now is giving LibOSmesa dependency issues, any ideas?

I've updated a package in AUR, bundling missing mesa libs into the bambu directly, should be working now.

guanyun-gudujian, 2025-06-04:
anyone who have the same black screen issue, pls help to check if follow package can work on your device, thanks
ubuntu 22.04:  https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091
ubuntu 24.04:  https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652384

attention pls: remove all the workaround first before testing these packages

GuDuJian-J-Zhang, 2025-06-09:
> > anyone who have the same black screen issue, pls help to check if follow package can work on your device, thanks ubuntu 22.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091 ubuntu 24.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652384
> > attention pls: remove all the workaround first before testing these packages
> 
> working on 25.04, nvidia & wayland

thanks for your feedback.
any garbage(tearing) rendering issue occurred?

3cl1ps, 2025-06-09:
> > > > > anyone who have the same black screen issue, pls help to check if follow package can work on your device, thanks ubuntu 22.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091 ubuntu 24.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652384
> > > > > attention pls: remove all the workaround first before testing these packages
> > > > 
> > > > 
> > > > working on 25.04, nvidia & wayland
> > > 
> > > 
> > > thanks for your feedback. any garbage(tearing) rendering issue occurred?
> > 
> > 
> > the preview was full of blacks artifacts that went away after slicing, i'll make a screenshot if that happen again.
> > edit : here : [image]
> > still have those errors :
> > ```
> > (bambu-studio:20989): Gtk-CRITICAL **: 08:09:17.670: gtk_widget_set_size_request: assertion 'width >= -1' failed
> > (bambu-studio:20989): Gtk-CRITICAL **: 08:09:17.670: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > but i don't think they're hurting.
> 
> yes that's it
> 
> can updating graphics card driver fix this issue?

latest prod, Driver Version: 570.153.02     CUDA Version: 12.8
i won't update to beta

guanyun-gudujian, 2025-06-10:
> latest prod, Driver Version: 570.153.02 CUDA Version: 12.8 i won't update to beta

ok

i understand

badnetmask, 2025-07-31:
For the record: Fedora 42, NVIDIA 575.64.03, Bambu Studio 2.1.1 via Flatpak
All I needed was to set this (either via --env or Flatseal): `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/glvnd/egl_vendor.d/50_mesa.json`
But honestly, I don't understand how it worked. This file does not exist anywhere on my system, and if I point to the file that actually exists (`/usr/share/glvnd/egl_vendor.d/50_mesa.json`) it does not work.
Hopefully the next version (2.2.0) will fix everything.

guanyun-gudujian, 2025-07-31:
> For the record: Fedora 42, NVIDIA 575.64.03, Bambu Studio 2.1.1 via Flatpak All I needed was to set this (either via --env or Flatseal): [...]
@badnetmask 
pls help to check which one work, thanks

version1 : (without platform checking)
for ubuntu 24.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366488248/artifacts/3562148808
for ubuntu 22.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366488248/artifacts/3562130141

version2: (with platform checking)
for ubuntu 24.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366503233/artifacts/3562161529
for ubuntu 22.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366503233/artifacts/3562154867

guanyun-gudujian, 2025-08-11:
> > For the record: Fedora 42, NVIDIA 575.64.03, Bambu Studio 2.1.1 via Flatpak All I needed was to set this (either via --env or Flatseal): `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/glvnd/egl_vendor.d/50_mesa.json` But honestly, I don't understand how it worked. This file does not exist anywhere on my system, and if I point to the file that actually exists (`/usr/share/glvnd/egl_vendor.d/50_mesa.json`) it does not work. Hopefully the next version (2.2.0) will fix everything.
> 
> This is even weirder, running on NixOS 25.05, there is nothing under that directory. It works, but I assume it somehow forces the app to look for the correct directory or run alternatively?

do not use this workaround

try the latest offical package pls

thanks

here pls: https://github.com/bambulab/BambuStudio/issues/4669#issuecomment-3162063639

Knifa, 2025-08-13:
> But honestly, I don't understand how it worked. This file does not exist anywhere on my system, and if I point to the file that actually exists (`/usr/share/glvnd/egl_vendor.d/50_mesa.json`) it does not work.

It gets mounted into the Flatpak environment from the Mesa Flatpak dependency. 

```shell
$ flatpak run --command=bash com.bambulab.BambuStudio
[~]$ ls /usr/lib/x86_64-linux-gnu/GL/glvnd/egl_vendor.d/
10_nvidia.json  50_mesa.json
```

snoack, 2025-08-14:
I just started experiencing the same issue. I don't have NVIDIA graphics, but only Intel integrated graphics. I'm on Wayland but the issue persists when switching to Xorg.

ZiqpL, 2025-08-14:
> [@Buom01](https://github.com/Buom01) [@badnetmask](https://github.com/badnetmask) [@3cl1ps](https://github.com/3cl1ps)
> 
> Hi, the offical package with the x11 mode workaround has been released
> 
> pls have a try
> 
> thanks
> 
> https://github.com/bambulab/BambuStudio/releases
> 
> attention pls: use the ubuntu version not fedora <img alt="Image" width="1496" height="511" src="https://private-user-images.githubusercontent.com/188939216/475289804-2246a34c-5d21-4a1f-9ad9-a30bfea3e8d2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTUyMDQyNDMsIm5iZiI6MTc1NTIwMzk0MywicGF0aCI6Ii8xODg5MzkyMTYvNDc1Mjg5ODA0LTIyNDZhMzRjLTVkMjEtNGExZi05YWQ5LWEzMGJmZWEzZThkMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgxNFQyMDM5MDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YmY1ODk4MjVhYTMyMGE5NDNjMzRmNTllZjFlNzQ4MWRlMjM1OGJhYjM3YzhmZmE2OGFjNTBhMzljMmQzZWZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.AydmQEnWenQFPESP48oivf99x5p0c_jrfgVQt0Yu1_k">

Thank you! This works perfectly for me. I’m using the Ubuntu version on CachyOS (Arch) with the following setup:

OS: CachyOS x86_64
Kernel: 6.16.0-7-cachyos
DE/WM: KDE Plasma 6.4.4 / KWin (Wayland)
CPU: Intel i5-13600KF
GPU: AMD Radeon RX 6700 XT
Memory: 31 GB
Display: 2560x1440 @ 165 Hz

Thanks again! It’s running smoothly on this setup :)

[3 more replies]