bambulab/BambuStudio #4671: System Filaments  blank window

WMate03, 2024-08-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Ryzen 5, 8GB RAM

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Select filament setting.

### Actual results

Nothing will appear at the system filaments.
[image]

### Expected results

I would select filament vendors.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16732587/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bisebaert, 2024-08-25:
Same problem on Windows 11 (Bambu Studio 1.9.3.50, P1S, Ryzen 5, 32GB).

realzorro, 2024-09-03:
We are aware of this issue and have fixed it. Please download the latest installation package from the official website to solve your problem.
Sorry for the inconvenience caused to you earlier. Please continue to follow and assist us in improving together.