bambulab/BambuStudio #4672: Color slicing is incorrect

GCRVL, 2024-08-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Linux Mint 21.3

### Printer

P1S

### How to reproduce

Slice a complex color model

### Actual results

Color is printed where no color should be.

[image]
[image]
[image]
[image]

### Expected results

Color should be printed only where needed.

### Project file & Debug log uploads

[broken_slicing.3mf.zip](https://github.com/user-attachments/files/16732677/broken_slicing.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

GCRVL, 2024-08-23:
For clarity, the screenshots show top layers that are white in the model, but are sliced being blue.

zhimin-zeng-bambulab, 2024-08-26:
[image]
It may cause some slicing errors, when there are non-manifold edges in the model.

closed 2024-09-05 by GCRVL