bambulab/BambuStudio #4673: Bambu Studio crashes when slicing model

AshleyVazCo, 2024-08-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.6.1

### Printer

Bambu Lab A1

### How to reproduce

1. Open project
2. Set up project. I used the cut tool to add connectors. Also used the text tool. Infill at 0%, walls at 4. The inside of the model is hollow.
3. Click Slice all.

### Actual results

The program crashes and a report is generated.

### Expected results

G-code should've been provided in order to print.

### Project file & Debug log uploads

[errorReport.zip](https://github.com/user-attachments/files/16732738/errorReport.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

AshleyVazCo, 2024-08-23:
The issue is with the text in the model. I took away the text and I was able to slice it. I added the text to the model in the slicer.

zhimin-zeng-bambulab, 2024-08-27:
Thanks for reporting this issue, we will try to fix it.

zhimin-zeng-bambulab, 2024-10-18:
@AshleyVazCo  It has been fixed in [1.10.0 Public Beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74), please have a try.

closed 2024-10-18 by zhimin-zeng-bambulab