bambulab/BambuStudio #4674: PVA between raft and model base

sealand-design, 2024-08-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	12th Gen Intel(R) Core(TM) i9-12900H   2.50 GHz
Installed RAM	32.0 GB (31.7 GB usable)
NVIDIA GeForce RTX 3080 Ti

### Printer

X1 Carbon

### How to reproduce

[image]

### Actual results

No matter what I try there is a layer of PVA between the raft and the base of the model -- so the model slips

### Expected results

there should be no PVA between the raft and the model base

### Project file & Debug log uploads

[PVA_btwn_raft_and_model.zip](https://github.com/user-attachments/files/16732848/PVA_btwn_raft_and_model.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-08-29:
Not a bug. The layers of PVA are raft interfaces. If you don't want it, you should disable raft.

sealand-design, 2024-08-29:
Okay, so there’s no way to tell it to do raft and supports but only PVA at where it interfaces with the actual model?

Is it possible to print something like what I’ve sent where the PVA between the raft and model base adheres well/becomes integral or does it always slip?

Thanks!

John

John

John O’Keefe, RLA
Principal
[cid:cb17999f-bd18-4718-8637-36355036971d]
236 Westminster St. Unit 502
Providence, RI 02903
(401) 649-6751

www.sealand.design<http://www.sealand.design/>