bambulab/BambuStudio #4679: Some text in dark mode not readable

capull0, 2024-08-24:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.6.1

### Additional system information

Apple MacBook Pro
Chip: Apple M2 Pro

### Printer

Bambu Lab P1S

### How to reproduce

1. Select "Dark Mode" in MacOS system settings
2. Open "BambuStudio"
3. for example open Device->Updates
4. Model, Serial, Version is colored in a very dark gray on a dark gray background
5. Open a shell/terminal and enter following to switch only BambuStudio to light mode:
`defaults write com.bambulab.bambu-studio NSRequiresAquaSystemAppearance -bool yes`
6. Open BambuStudio to see, that parts are now white, but most is still in dark gray and still not readable.
7. delete this setting with:
`defaults delete com.bambulab.bambu-studio NSRequiresAquaSystemAppearance`
8. select light mode in macOS system settings and everything is readable.
9. Setting "dark_color_mode" to 1 in BambuStudio.conf will be overwritten, after next start

### Actual results

[image]
[image]
Actually the text is on the screenshot a bit better readable, but not inside BambuStudio, even you put your eyes close to the monitor.

### Expected results

1. everything should be readable
2. if i use "NSRequiresAquaSystemAppearance" to set the mode explicit for BambuStudio, the app should follow this mode

Just change the color to a brighter gray :-)

### Project file & Debug log uploads

there are no logfiles, because issue shows on a fresh installation.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

capull0, 2024-08-24:
it seems like there is a setting in macOS, which is forcing this color problem.
Apple menu > System Settings > Accessibilty > Display
If you disable "Increase contrast", the text is readable.

but i want to keep this setting enabled 😞

walterwongbbl, 2024-08-29:
@capull0 
I think I understand your request, and I will put it in the requirements pool for further discussion.

pinheiroalexandre2, 2024-09-15:
This happens in linux to. 

[image]

foxsam21, 2024-10-30:
I get this with `Bambu_Studio_ubuntu-20.04_PR-5005.AppImage`, not in dark mode.
[image]

MackBambu, 2025-03-26:
Fixed in v2.0

bbrendon, 2026-04-13:
@MackBambu it doesn't appear to be fixed. I've been dealing with invisible text for as long as I can remember.