bambulab/BambuStudio #4680: Supports are not created inside an object when shell layer height is 0

flyandi, 2024-08-25:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

No performance issue

### Printer

P1S

### How to reproduce

Load an object with internal structure and enable supports. 
Set Top and Bottom Shell Layers to 0.
Slice it.

### Actual results

No supports are created inside the open object.

[image]

### Expected results

I expect the slicer to understand that the object is open on both sides and place support structures inside of the object.

### Project file & Debug log uploads

General issue

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-08-29:
The slicer isn't that smart now... But it's a good suggestion. We'll consider adding the feature in the future.

flyandi, 2024-08-29:
Thank you for responding. That is good to know that the slicer doesn't understand that yet. I probably have to design internal structures that I can break of myself for now.