bambulab/BambuStudio #4681: Text font

Jafterdark, 2024-08-25:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macos 14.6.1

### Additional system information

MacBook Pro

### Printer

x1 carbon

### How to reproduce

I have some custom fonts when I select them they just default to a non-custom font anything cursive seems to go back to standard or script looking

### Actual results

[image]

### Expected results

Pretty much pick anything that script looking cursive looking seems to not work if I go under OrcaSlicer, this works fine if that helps

### Project file & Debug log uploads

I can include any files because probably don't have a font

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-08-26:
Hello, can you record a GIF? Thank you

MackBambu, 2024-09-01:
hi,@Jafterdark 
Can you provide the font files ? I've been working on this problem recently and need to test case.

Jafterdark, 2024-09-01:
Here's a screenshot that shows the font and how it looks in the pulldown
menu. When you select the font, it changes to a standard font, even in the
menu where you select the font.

[image: image.png]
Hope this helps

cdubs-cyber, 2024-11-04:
I just submitted the same bug... I probably should have searched first. https://github.com/bambulab/BambuStudio/issues/5137

cdubs-cyber, 2024-11-04:
This bug also appears to be the same issue: https://github.com/bambulab/BambuStudio/issues/4992

abandonedbrain, 2024-11-29:
Confirmed I'm seeing this as well, and not with every font. All will default to Arial (Windows version of Bambu Studio, running on Win 11). Test font installed with 'Install for all users' option, shows in list but when placed it's Arial.

Workaround: Do your design work in Orca Slicer (which IMHO has better text editing than BS), save the project, then open it in BS. You'll still have an object layer with the type, but as paths (no longer editable text).

th3tick, 2025-01-24:
I just downloaded a TTF I wanted to use, but it does exactly as described here.  So I guess this is still broken.