bambulab/BambuStudio #4690: Camera Shuts off and States "Temporarily closed because there is no operating for a long time."

Gervin33, 2024-08-26:
### Bambu Studio Version

1.9.4.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Leave camera running

### Actual results

After a while the camera goes black and a message appears stating "Temporarily closed because there is no operating for a long time."  
[image]

### Expected results

The camera should continue running until I hit stop.  If this is a new built in feature, there should be a preference setting to not have this happen as I like to keep the camera running to monitor as I am working at home.

### Project file & Debug log uploads

No log or project file required, works with all file/projects.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Spyder96T, 2024-08-27:
Same complaint here. Just started with the new studio update.

lorissimo, 2024-08-27:
Dear Bambu... Fix this. Kind regards.

Spyder96T, 2024-08-27:
Found a workaround until Bambu fixes this.. the message appears when no action is taken in the slicer. I added a mouse move .exe file from the windows store and set it to do a left mouse click every 30 seconds. Had the camera running uninterrupted for 30 minutes now

tinman31k, 2024-08-27:
Same issue as above.  Did not happen until installation of new version in Studio.

NoUniqueNameFound, 2024-08-27:
Please add a timeout value for the preferences tab, or provide an option to disable it?
I often observe the progress of prints, and this "feature" is annoying.

lanewei120, 2024-08-27:
Hello~

Thanks for reporting this issue
Let me give some background information here:

Many of our users have previously complained that Bambu Studio prevents their PCs from entering sleep mode. Considering the need for energy saving and reducing data costs, this version introduces an automatic pause for videos after a long period of inactivity. Taking into account some users' need for continuous remote video streaming, we will consider other optimization methods in future updates to better accommodate the needs of these users.

lanewei120, 2024-08-29:
Hello
@lorissimo @robo-1915 @tinman31k @Gervin33 @NoUniqueNameFound @Spyder96T 

could you show us your liveview scenario?
and how about following logic:
if the printing is ongoing, we can keep the liveview on
and if the printing finish, the liveview will be paused if there is no operations for a long time.

Gervin33, 2024-08-29:
My scenario is that I work from home and sometimes have longer prints that I feel more comfortable if I can see the print still running.  I often have it open on a second screen while my work computer is on the other screen.

As far as your logic, that would work for me.  As long as the live view stays on while the printing is happening, it would work for me.  An option to disable the automatic shutoff would be nice, but not 100% mandatory in my case.

NoUniqueNameFound, 2024-08-29:
That logic sounds good to me.  If the printer is idle, I’m OK with the camera going idle.  

When the printer is active, I want the camera to have the ability to run without a timeout.

 

On another note, It would be nice if there was a way to turn off the chamber light when a print finishes - Is there a G-code value for the chamber light, or could toggling that be incorporated somehow?

robo-1915, 2024-08-29:
I would be perfectly fine if the camera timed out when the printer is idle, just not when it's running. It would be nice if these were options though.
Camera timeout after 15 minutes when printer active = on/off
Camera timeout after 15 minutes when printer idle = on/off

longrackslabs, 2024-08-30:
This is bad change.  Users don't need you deciding how long the video plays.  What if it is on a monitor-only setup ?  

I would love to know the origin of this bug/change.  Who's idea was it, who thought was a good idea, and how it was approved.

lorissimo, 2024-08-30:
I wholeheartedly agree that this is a horrible change. We should be able to turn the stream on or off however we see fit. We shouldn't be mandated with a prefixed unchangeable timeout. At the very least, options would have been helpful versus none at all.

mysterious-roman-dodecahedron, 2024-08-31:
I don't like this either. I don't want the camera feed to be disabled, even if the print has completed. 
As both behaviors (timeout and no timeout) have been programmed, please consider adding a simple check box in the options menu to enable/disable this behavior. I would like to retain the previous functionality.

hervema, 2024-09-07:
The new behavior would be acceptable if users can change the default setting in preferences for those who wish to do so. It should not be enforced! Thanks for your support, Bambu team! :-)

robertwallace, 2024-09-08:
I've got a quick and so-far permanent fix for this problem:  Switch over to Orca Slicer.  My camera feed hasn't "temporarily closed" once when using Orca.    Bambu is touted as being a top-notch user-experience company.  Not so much, for this issue, it seems.

KedziePanda, 2024-09-09:
Why is this even a discussion, there was no reasonable reasoning to ruin the functionality of the camera monitoring? Please revert the change or make it a choice, Thanks.

lanewei120, 2024-09-09:
yes, we will add an option in next version

thanks all for the feedback and suggestions

longrackslabs, 2024-09-09:
And the default should be the previous behavior - the setting should enable
the timeout

tinman31k, 2024-09-09:
Thank you!

Links:
------
[1] 
https://github.com/bambulab/BambuStudio/issues/4690#issuecomment-2338454476
[2] 
https://github.com/

blue68camaro-git, 2024-09-12:
I turn on my printer, let it boot up. Then open Studio. Import an .stl and slice and print. I receive this message. Studio has not been even open for over 14 hours. MacOS version.

UltimateCodeWarrior, 2024-09-13:
So I bought the X1C with the enclosure so I could print ASA (toxic) way away from me.   I liked the X1C because it had an integrated camera, carbon filter, enclosure. Wow!     What good is all that if the camera shuts off half way through the print, now I have to walk down 2 flights of stairs and over 60' away to look into the printer to make sure it's not spaghettied up.     Not good.       

That Error Message!        All Your Base Are Belong To Us.  

Now what I don't get is when you go to Bambu Labs website to report an issue.    All I should have to do is press a button in Bambu Labs saying something is wrong.     Then it's got logs, times, serial numbers, videos, make files, stl, recorded video,  dates and times.   There is absolutely ZERO reason why any of us should need to fill out a form of any kind.      Hell, a chat should be embedded into  bambu labs, not only a support agent chat, but also a user chat because truth be told, I have found more help on reddit and youtube than filing the complain/form/bug/issue on BambuLabs.     And I don't want some wonky work-around, I want a nice clean integrated solution that just flows and works the first time around.        Bambu labs should get 10 people off the street pay them to go through a setup and first print, and then revise their documentation complete with videos on trouble areas, and do that process over and over again until there are no more show stoppers or funkyness during it.  That's how they will achieve a #1 status.

eoosting, 2024-09-16:
If users would like their computer to sleep, turn off the live view. I actually really appreciated the fact that liveview prevented my computer from sleeping when video was running. If my computer went to sleep while watching youtube or netflix it'd be really anoying. I presume that most everyone with live view turned on wants it to stay on, and they have the option to turn it off if they desire. Please revert, or if not, put in a time out value that could be made long enough as well as disabled.

k thx

eoosting, 2024-09-16:
> Hello~
>
> Thanks for reporting this issue Let me give some background information here:
>
> Many of our users have previously complained that Bambu Studio prevents their PCs from entering sleep mode. [...]

If users would like their computer to sleep, turn off the live view. I actually really appreciated the fact that liveview prevented my computer from sleeping when video was running. If my computer went to sleep while watching youtube or netflix it'd be really annoying. I presume that most everyone with live view turned on wants it to stay on, and they have the option to turn it off if they desire. Please revert, or if not, put in a time out value that could be made long enough as well as disabled.

Also, just fyi, it seems like my computer doesn't even go to sleep once the video has timed out. Please put it back.

ckcampbell248, 2024-09-17:
Adding my vote to make this an option rather than making a unilateral decision to shut if off for everyone. It is really annoying that the video feed shuts off. I generally have it open on a separate computer from the one I am working on, and I frequently glance over to make sure things are progressing successfully. This change completely negates the ability to do that.

johnnybravo720, 2024-09-18:
First, I'd like to know how this affects timelapse. Is that being shut down when camera goes blank? Also, many of us keep our printers in another location, and this continuous monitoring is vital to assessing quality of prints. 

My understanding is that the camera does not go into play mode unless the user manually hits play. So if a user wants their computer to enter sleep mode, that user can also manually stop play. 

Think about it as a person going to sleep in their bed.  Either that person needs to hit a switch to turn off those lights, or program a device (such as amazon  echo) to do it.  Bambu lab is accommodating the person that can't just hit the switch themselves. It makes no logical sense why this unpopular feature has been pushed on us. 

So I'm going to be one of the users requesting this feature be removed. At the very least, give us control of our own switch.

hervema, 2024-09-19:
Hello. The last version (1.9.7.50), just released, add the switch option "keep liveview when printing" in the preferences and it works for me perfecty!  We now have the choice, perfect ! Thanks a lot for that, Bambu team! :-)

longrackslabs, 2024-09-19:
This is better, but shouldn't it default to on ?

longrackslabs, 2024-09-21:
but it looks there is still part of this feature you can't shut off "Streaming stopped because printing has stopped for a while"

How can I be more clear - its not your decision!

robertwallace, 2024-09-22:
Installed the update.  **They still don't understand** the problem.   Yes, you can set the liveview option on and the feed does stay on during the print, but when the print is complete, the old behavior returns - the video feed times out and is cut off.  The video feed should not cut off "automatically" ever.

longrackslabs, 2024-09-22:
100%.   This is a great cast of "fixing" a "problem" for one user, that
causes many problems for many other users as well as
unintended consequences.  If I were them (and I am not), I would back out
the change completely and start over with a really good use case / user
story for the video streaming / monitoring epic/feature set.  The way they
did it just seems so hacky and amatuer.

sperman32, 2024-09-23:
I agree with those above that want the camera on 100% of the time.  Please change back to the original behavior.

robo-1915, 2024-09-23:
For the one or 2 users that wanted their computer to go to sleep, all they had to do was stop the video! This was never something that needed to be 'fixed'.

longrackslabs, 2024-09-23:
agreed.  or it was a microsoft windows specific fix to honor the system
setting that was shown.

RazaXML, 2024-10-03:
> For the one or 2 users that wanted their computer to go to sleep, all they had to do was stop the video! This was never something that needed to be 'fixed'.

Exactly what I was thinking.  There's no way more than a few people actually had this "complaint"  I understand that they may not have known why it was keeping their computer from sleeping, but the solution should have been a checkbox in the preferences for "Allow windows to enter sleep mode" (which does the whole video timeout thing)  that's it...

ttt888zzz, 2024-10-11:
1.Please download the latest version and visit the website，https://bambulab.cn/zh-cn/download/studio
2.Update 【References】setting，according the photo
[image]
https://github.com/bambulab/BambuStudio/releases

closed 2024-10-11 by ttt888zzz

KedziePanda, 2025-05-17:
Yea it’s still not fixed, still times out. They are just salty about the bandwidth usage since they insist on monitoring the cameras from china, there are some workarounds bypassing their ability to even see the printer and continue to use it. As usual manufacturers decide to act like they still own the equipment the customer paid for, so plenty of us just remove their ability to do anything with our equipment and disable their access and stream the camera locally or with our own servers. Sure the Ai features of print failure detection aren’t capable of being used but the majority of those with specific use cases don’t care and would have gladly accepted the option to disable that feature and their ability to monitor both sound and video whenever they wish, instead of timing out the stream.

Still some of the best hardware and ecosystem for simplicity and reliability, just needs a little less overlord of a company sticking their noses where they don’t belong.