bambulab/BambuStudio #4695: ams not sinking

Keving1950, 2024-08-27:
### Bambu Studio Version

1.9.4.60

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

12 gen intel

### Printer

A1

### How to reproduce

1 add new filament
2 shows up on device tab
3 syn or resync in prepare tab not updated, restarted Studio and printer will not update prepare tab, print tab shows correct filament position
[image]
[image]
[image]

### Actual results

even wrong if left overnight

### Expected results

want prepare tab to update

### Project file & Debug log uploads

any file
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DanBao-Bambu, 2024-09-05:
May I ask if you have clicked the button and performed a resync. Also, please note that if multi device management is enabled, the sync mechanism will not take effect.

Keving1950, 2024-09-05:
seems to have fixed itself, I I think when I have A1 and Mini connected at
the same time may be cause
Kevin Guest
0412874400

DanBao-Bambu, 2024-09-06:
Thank you for your reply. This question will be closed, and please feel free to ask any further questions.

closed 2024-09-06 by DanBao-Bambu