bambulab/BambuStudio #4696: macOS crash when sending print job

iCMDdev, 2024-08-27:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1 Developer Beta 1 (build number 24B5009l)

### Additional system information

SoC: M1 Pro

### Printer

Bambu Lab P1P

### How to reproduce

Potential duplicate of #4600 and / or #4612. Note that the device was on a beta build of macOS Sequoia, so that could also cause it.

Bambu Studio crashed while sending a print job to the printer. It's pretty hard to replicate this issue, as it occurs quite randomly. I suspect this might be related to some GUI race conditions.

### Actual results

Shouldn't crash, I guess :smile

### Expected results

Crashes (quite randomly) when sending a print job.

### Project file & Debug log uploads

I attached the macOS-generated crash log file.
[crashreport.log](https://github.com/user-attachments/files/16759885/crashreport.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

iCMDdev, 2024-08-27:
The print job was successfuly sent to the printer even though the app crashed, so I suspect the crashed occured when switching to the camera feed.

TinyMito, 2024-08-28:
I currently have a ticket opened with Bambu; I haven't got the time yet to record a video as they requested.

Ticket No: US240827785002

walterwongbbl, 2024-09-04:
@iCMDdev 
Thank you very much, we are trying to solve this issue.

walterwongbbl, 2024-09-04:
@iCMDdev @TinyMito 
Hi. We have 3 Mac books, all upgraded to 14.6.1 But we haven't found this issue, can you provide Studio's logs

iCMDdev, 2024-09-04:
> But we haven't found this issue

I can't really reproduce this issue either, to be honest. As I said, it occurs quite randomly 😢

> can you provide Studio's logs

Sure: [debug_Sun_Aug_25_11_57_19_17928.log](https://github.com/user-attachments/files/16867396/debug_Sun_Aug_25_11_57_19_17928.log)