bambulab/BambuStudio #4701: Live timeout to be optional

spadger1066, 2024-08-28:
Version 1.9.4.60 has introduced a fixed 15 minute timeout of the live feed.

I would like either this to revert back to previous versions where the live feed was continous or at least have the option to make it so.

Only option appears to manually refresh the feed every 15 minutes

LightDemon, 2024-08-29:
This change is incredibly annoying. The whole point of having a camera in a $1500 printer is to be able to keep an eye on the print. Why must I stop what I'm doing to click a screen every 15 minutes to keep the video feed running?

longrackslabs, 2024-08-30:
+1 to this.  not sure how or how this decision was made, but its bad.

FatHammer, 2024-09-01:
Fully agree with this. It should be an option in the settings, rather than being forced.

dbosman, 2024-09-02:
Should indeed be optional and/or have a setting where you can select the timeout duration. This makes it difficult to monitor.