bambulab/BambuStudio #4704: New Updates have issues importing from makerworld directly

LawnBurgundy, 2024-08-28:
### Bambu Studio Version

1.9.4.60

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 enterprise

### Additional system information

32.0 GB (31.7 GB usable, 13th Gen Intel(R) Core(TM) i7-13850HX   2.10 GHz

### Printer

X1E

### How to reproduce

1. go to makerworld
2. select any file to "open in Bambu studio" 
3.  Watch it not open correctly and time out

### Actual results

[image]
[image]

### Expected results

file should import.

### Project file & Debug log uploads

[Uploading log.zip…]()

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-08-29:
Hello, can you provide 3mf

LawnBurgundy, 2024-08-29:
It’s any file from makerworld directly to open In Bambu studio. I can download the stl and import but the direct open is what is not working Sent from my iPhone

ttt888zzz, 2024-08-30:
HI，[LawnBurgundy](https://github.com/LawnBurgundy)：
Could you Please provide a video，we will check it

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

GraffitiDecos, 2024-11-10:
Hey, I just downloaded Bambu studio for the first time and ran into the issue. Since I don't yet have a printer I didn't install the network adapter. I decided to try that and it fixed the issue.

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00