bambulab/BambuStudio #4708: Adaptive Layer Height causing issues with Tree supports

NXSmiggy, 2024-08-28:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1s

### How to reproduce

Slice file with adaptive layer heights on and tree supports enabled

### Actual results

When slicing the supports are glitched out and through the model. Same mode sliced on earlier vetsion this week was working fine. File sliced at 0.2mm height fine, adaptive layers used on top layers for smoother curves. Adaptive layer slider also seems to have high integer value and takes a lot longer to increase from 0.2 to 0.8 then previous versions
[image]

Without adaptive layers
[image]

With Adaptive Layers
[image]
[image]

### Expected results

Tree supports are generated cleanly

### Project file & Debug log uploads

[Uploading Adaptive Layer Issues.zip…]()

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-08-29:
It's not a bug. When adaptive layer height is off, default tree style is "organic tree"; When it's on, default is "hybrid tree". Wall count is 0 for hybrid tree, so you saw the exposed infills. You should be able to print the file without any problem.
If you don't like hybrid tree, you can change other styles like "slim tree", which can also be used with adaptive layer height.

NXSmiggy, 2024-08-29:
> It's not a bug. When adaptive layer height is off, default tree style is "organic tree"; When it's on, default is "hybrid tree". Wall count is 0 for hybrid tree, so you saw the exposed infills. You [...]

When trying to print the file the supports were open in mid air and intersecting through the model so dont think this is expected behaviour.

[image]