bambulab/BambuStudio #4713: LAN Mode does not remember password after computer sleeps

CaptnJB, 2024-08-29:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 12.7.5 Monterey

### Printer

A1

### How to reproduce

Printer in LAN mode.
Add Printer using passcode
look at the device, stream video, whatever....
let computer sleep/reduce power.
click on devices, printer, it's a valid green printer, it will try to connect and then say incorrect password.  I might understand if you shut Bambu down, but just letting the laptop sleep shouldn't make it loose the password.

### Actual results

[image]
[image]
[image]
[image]
[image]

### Expected results

Not re-entering your local printer passcode every time.

### Project file & Debug log uploads

[Gridfinity Wall Rack 3mf.zip](https://github.com/user-attachments/files/16803365/Gridfinity.Wall.Rack.3mf.zip)
It doesn't matter what file you're in.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kevinvanblerk, 2024-09-08:
I am experiencing this exact same issue, also on my MacBook, also after MacBook sleeps, extremely frustrating.

jon-walton, 2024-09-09:
I had the same problem and noticed that in `~/Library/Application Support/BambuStudio/BambuStudio.conf` , there are 2 areas for access code and they didn't match.

```json
{
  "access_code": {
    "printer_id": "abc123"
  },
  "user_access_code": {
    "printer_id": "not_the_same"
  }
}
```

Close all instances of Bambu Studio, edit the file to make both access codes correct and it started working for me

kevinvanblerk, 2024-09-09:
> I had the same problem and noticed that in `~/Library/Application Support/BambuStudio/BambuStudio.conf` , there are 2 areas for access code and they didn't match.
> 
> ```json
> {
>   "access_code": {
>     "printer_id": "abc123"
>   },
>   "user_access_code": {
>     "printer_id": "not_the_same"
>   }
> }
> ```
> 
> Close all instances of Bambu Studio, edit the file to make both access codes correct (matching what the printer shows) and it started working for me

Thank you @jon-walton that worked

psjshep, 2024-09-17:
For me, on Windows, the 1st occurrence of the access code had a newline notation added to the end :

{
    "access_code": {
        "xxxxxxxxxxxxxxxx": "yyyyyyyy\n"
    },
    
 When I removed the "\n" the lan code was remembered... thx @jon-walton for finding the problem file

jiexunlijx, 2024-09-24:
How does a fix this simple, take this long to remain unfixed? And it wasn't even there to begin with, it started with one of the more recent updates. It used to work just fine

psjshep, 2024-09-24:
I just tried the latest update & there now doesn't seem to be 2 sections for the password... & the newline character is no longer saved...  try it to confirm....

walterwongbbl, 2024-10-11:
@psjshep 
Does this seem to only happen on macOS systems?

psjshep, 2024-10-11:
As per my original post, I am using Windows

phrxmd, 2024-10-12:
Seems to be the same issue as #4076. Fixing the mismatch between the two entries in `BambuStudio.conf` fixes it for me (on 1.9.7.52 on Linux).

psjshep, 2024-10-21:
> I just tried the latest update & there now doesn't seem to be 2 sections for the password... & the newline character is no longer saved... try it to confirm....

Correction :
After the quick-fix is carried out, the issue is resolved across version update.... but on a newly installed windows machine, the issue of a new line character is still present... editing as described to remove the newline char fixes the problem but the latest Studio is still creating the problem on configurations not quick-fixed

pleasantone, 2025-10-07:
https://github.com/SoftFever/OrcaSlicer/pull/8256