bambulab/BambuStudio #4715: Exporting plate sliced file saves file with .gcode.3mf.gcode.3mf

gary-wlg, 2024-08-29:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOs 14.5 Sonoma

### Additional system information

When exporting a plate sliced file, the suggested filename has a .gcode.3mf, but when you save it, an additional .gcode.3mf is added to the filename, sometimes it saves with ".gcode.3mf" repeated 3 times.

### Printer

Bambu Lab P1S

### How to reproduce

1. Open Bambu Slicer, add a primitive cube.
2. Slice file
3. Click Export Plate Sliced File

The suggested name of the file is Untitled.gcode.3mf.gcode.3mf

### Actual results

When you click Save you end up with a file called

Untitled.gcode.3mf.gcode.3mf.gcode.3mf

### Expected results

I would expect the file to be named Untitled.gcode.3mf

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/user-attachments/files/16805620/Cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

TNT3DPrintsLLC, 2024-10-10:
I am having the same issue.

Haidiye00, 2024-10-11:
https://jingyan.baidu.com/article/90808022d9dbbcbc91c80f9c.html you can close suffix names' display on mac.And we have currently resolved the storage name issue. Please wait for the next version~

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00

mchampanis, 2025-09-27:
This is still happening in Bambu Studio 2.1.1.52, macOS 14.7.6.

`Prepare` > `Slice plate` > Change `Print plate` to `Export plate sliced file` and click

File is saved with `.gcode.3mf.gcode.3mf` and complains if you try to correct it manually.

[image]

[image]

[image]