bambulab/BambuStudio #4717: Studio crashing when slicing (have sliced same objects several times before)

ksousa410, 2024-08-29:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

Cpu - i914900k, GPU RTX 4090, memory 48gb

### Printer

x1c

### How to reproduce

open bambu studio. load project. select preview. 

### Actual results

It slices to 15% and then bambu studio crashes/closes. Sometimes it will get as far as 50% before crashing. 

### Expected results

I would expect it to slice and generate a preview. My system far exceeds the system requirements, so would not expect any issues. I have sliced this same project twice this week without issue (recently purchased 3 more AMS systems, so I needed to print three of these same objects to complement the AMS). I tried several things which have worked in the past with other projects like moving the object, changing minor settings etc. I even tried re-installing the previous bambu lab version as this issue only arose after this last update, but that did not help so I re-installed the latest. 

### Project file & Debug log uploads

[bambu-studio.exe.21952.zip](https://github.com/user-attachments/files/16806007/bambu-studio.exe.21952.zip)
[Dry Pods with Round Hygrometer and Desiccant Trays.zip](https://github.com/user-attachments/files/16806022/Dry.Pods.with.Round.Hygrometer.and.Desiccant.Trays.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-08-30:
HI，[ksousa410](https://github.com/ksousa410)：
Please try the following steps，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

QingZhangBambu, 2025-05-23:
close for no reseponse

closed 2025-05-23 by QingZhangBambu