bambulab/BambuStudio #4719: Shutting down when launching print

menkheperra, 2024-08-30:
### Bambu Studio Version

19.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

MACOS Sonoma 14.6.1

### Additional system information

CPU M2 mem:256G Mackbook Pro 2024

### Printer

Bambulab X1C,P1S,A1mini

### How to reproduce

Impossible to reproduce because very random:
Many times, too much times, I slice the model (simple or complicated doesn't matter) and when i send to printer (via wi-fi but not lan-only) The software shuts down saying it unaspectately close and asking me to ignore or open again.
I bought printers on march and use this software since i bought them. This problem has been always present also on a new reinstalled clean copy of MacOS.

### Actual results

There is no screen shot to post. The software shuts down.

### Expected results

It asks me if restart the softwre or ignore the error.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16815516/log.zip)
[CubenGear.3mf.zip](https://github.com/user-attachments/files/16815523/CubenGear.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

buges, 2024-08-30:
I have exactly the same problem.

M2 Max Studio 512gb

fahaaaxiki, 2024-09-03:
@menkheperra Could you please provide the crash report？Thanks

menkheperra, 2024-09-03:
I already added a log file and the project file.
**Where do I find the "crash" report on my Sonoma ?
Must I wait the issue reproduce before sending crash report ? **

Anyway I've read many other posts with the same issue: studio keeps shutting down once print job has been sent to the printer (as someone said it seems at moment of switching to the device page).

walterwongbbl, 2024-09-10:
Hi. @menkheperra 
When a crash occurs on macOS, you will be prompted to open the crash report.