bambulab/BambuStudio #4723: Bambu Studio will not recognize SD card on brand new P1S

jsalegit, 2024-08-31:
### Bambu Studio Version

01.06.01.00

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Labs P1S

### How to reproduce

1.Turn printer on
2. View files on SD card via printer screen (works)
3. Open Bambu Studio and log in
4. Select Device (P1S)
5. Click MicroSD Card Option
6. Select Model Tab
7. Get File does not exist

### Actual results

Can't see any files on SD card.

### Expected results

See files on SD card

### Project file & Debug log uploads

don't have a project file or debug log.  Have not tried to print anything yet from Bambu Studio.  Just looking at app to see what it has, and noticed I can't access SD card. 
Also, I can't access SD via the Handy Android app either.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gaolegao-lx, 2024-09-03:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

slippystance, 2024-11-06:
Confirmed.  I just got my P1S and Bambu Studio (PC/Win11) is not seeing any files on the SD Card.  However, I can confirm that the P1S can see those files.
[image]

Sm0ku, 2024-11-22:
Same for me, just got a new P1S and exactly the same issue.

policeman0077, 2024-11-23:
Operating System: Win11 Pro 23h2
Bambu Studio Version: 1.10.1.50
Printer: P1S
Printer Firmware: 01.06.01.02

I tried LAN mode it was not working as well.
But I am able to connect to the printer via a FTP client. 
I could see all the models(gcode files) and time lapse in FTP client.

[image]
[image]
[image]

Sm0ku, 2024-11-23:
I tried copying all the files from the included SD card to a different card, same problem. Put in an empty card and after a print i can see the timelapse and the model of the print on the new SD card. Maybe one of the files on the original card is corrupted, will try investigating further when i have some time.

JevadRolls, 2024-12-08:
Same issue, Any resolution?

I just unboxed and set up the new P1S, I can see the card contents from the printer display. Printed the benchy from the included SD Card just fine. Both Bambu Studio & Handy App see the printer, can view video, change configurations, both Apps report same no SD card screen??

policeman0077, 2024-12-10:
> Same issue, Any resolution?
>
> I just unboxed and set up the new P1S, I can see the card contents from the printer display. Printed the benchy from the included SD Card just fine. Both Bambu Studio & [...]

Try reset the printer to factory restart and use older version of bambu studio might help. I was using the latest bambu studio 1.10. Then I found this problem and send a ticket to bambu. They guide me to a bunch of debug process and no luck. Then I reset the printer and installed bambu lab 1.9.7.52. Seems working for me for now.

Maddy1206, 2025-02-24:
Just unpacked my P1S and still the same problem. Any update from Bambu on this part?

juanjuanmejias, 2025-03-21:
I confirm I have the same issue with P1S just out of the box (after firmware update). Have there been any news?

svander01, 2025-04-11:
Same issue here, brand new printer. Any help?

glibvoronin, 2025-06-06:
Can confirm the same on a fresh P1S.
Firmware: 01.08.02.00

DNKROZ, 2025-06-30:
Just got my new P1S with AMS and I've been scratching my head for a few minutes before cheking this thread, thanks a lot for reporting the issue.

Firmware: 01.08.02.00 (last version), can't see the files from the app or the PC but SD card works fine and models are there and can be printed from the SD card