bambulab/BambuStudio #4728: Video stops

tomt47, 2024-08-31:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 pro

### Printer

Bambu X1 Carbon

### How to reproduce

Start video under device tab after a few minutes video shuts off. Can restart but video only stays on for a few minutes.

### Actual results

Video shuts offf

### Expected results

Hopefully you can patch this

### Project file & Debug log uploads

[debug_network_Thu_Aug_29_22_58_23.log.zip](https://github.com/user-attachments/files/16826165/debug_network_Thu_Aug_29_22_58_23.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ttt888zzz, 2024-09-02:
https://github.com/bambulab/BambuStudio/issues/4690
This could be beneficial for you.

ttt888zzz, 2024-10-11:
1.Please download the latest version and visit the website，https://bambulab.cn/zh-cn/download/studio
2.Update 【References】setting，according the photo
[image]
https://github.com/bambulab/BambuStudio/releases

closed 2024-10-11 by ttt888zzz