bambulab/BambuStudio #4729: Wall Splits

topheryorks, 2024-08-31:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.6.1

### Additional system information

Apple M1 Pro
16GB ram

### Printer

Bambu P1P

### How to reproduce

slice file using .12mm Fine @BBL P1P preset with Arachne wall generator.
print file on Bambu Lab P1P
cookie cutter file is having wall splits

### Actual results

We are a cookie cutter company and have been using the same .12mm Fine @BBL P1P setting (with Arachne wall generation) since January of 2024.  Great results each time.  

After the recent Bambu Studio update we have been having wall splits.

I notice the new presence of gap infill on our slices which didn't used to appear.  
[image]

### Expected results

Cutter file printed without wall splits.

### Project file & Debug log uploads

[debug_network_Sat_Aug_31_16_48_14.log.enc.zip](https://github.com/user-attachments/files/16826304/debug_network_Sat_Aug_31_16_48_14.log.enc.zip)
[BambuStudio.conf.zip](https://github.com/user-attachments/files/168263
[Cute Chubby Cat 3.75.stl.zip](https://github.com/user-attachments/files/16826307/Cute.Chubby.Cat.3.75.stl.zip)
05/BambuStudio.conf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

topheryorks, 2024-09-02:
[Archive.zip](https://github.com/user-attachments/files/16838093/Archive.zip)
 attaching the first two files with problems after the upgrade to 1.9.5.51

QingZhangBambu, 2024-09-03:
@topheryorks hi, the file you send use classic mode, so it had gap. the arachne mode looks fine.
[image]
[image]

closed 2024-10-12 by QingZhangBambu

topheryorks, 2024-10-12:
Thanks.  The most recent Bambu studio update fixed the problem (1.9.7.52)