bambulab/BambuStudio #4731: Crashes on upload of Print

daggyPants83, 2024-09-01:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.6.1 (23G93)

### Additional system information

MacBook Air M2, 2022, 24GB RAM

### Printer

Bambu P1S

### How to reproduce

Every single time I press "print" it completes the progress bar for uploading, and crashes the whole program the moment after completing the upload.

### Actual results

Bambu Studio is no longer running, and the apple bug report prompt comes up.

### Expected results

The application should not crash

### Project file & Debug log uploads

[actuator limit switch.3mf.zip](https://github.com/user-attachments/files/16826824/actuator.limit.switch.3mf.zip)
There is no log file for the crashes

### Checklist of files to include

- [ ] Log file
- [X] Project file

fahaaaxiki, 2024-09-03:
@daggyPants83  Could you please provide the crash report？Thanks

daggyPants83, 2024-09-03:
Here is one of the Mac OS logs:

￼

fahaaaxiki, 2024-09-03:
> Here is one of the Mac OS logs: ￼
> […](#)

Sorry，it seems like upload failed.Maybe you could upload on Github via “attach files”
[image]

walterwongbbl, 2024-09-10:
@daggyPants83  Thank you very much.