bambulab/BambuStudio #4734: Cierre inesperado

Luisdjnas, 2024-09-01:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 profesional

### Additional system information

I9 14900k, Msi mag z790 tomahawk wifi, 64gb ram ddr5, msi gtx4070 12gb

### Printer

Bambulab X1 carbon

### How to reproduce

Introduzco stl de gran tamaño en este caso una litofania, elijo perfil, intento laminar

### Actual results

El programa se cierra solo

### Expected results

El resultado tendría que ser el archivo gcode

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16827922/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-09-02:
hola  [Luisdjnas](https://github.com/Luisdjnas) :
Por favor, pruebe los pasos，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

HI [Luisdjnas](https://github.com/Luisdjnas) :
Please try the following steps，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

QingZhangBambu, 2024-10-12:
@Luisdjnas if you still can not slice the model after set CPU settings, upload 3mf files. i will check

Luisdjnas, 2024-10-13:
If changing the CPU affinity did cut the file correctly. The bad thing is that every time I close the bambustudio program I have to change it again. I can't leave the affinity fixed for bambustudio