bambulab/BambuStudio #4735: Updated Software Version

Infinity-Papa-Joe-9mm, 2024-09-01:
### Bambu Studio Version

1.9.4.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Version 22H2 (OS BUILD 22621.4037

### Printer

Bambu Lab P1S

### How to reproduce

[image]

### Actual results

Ever since updating to the newest version, I continue to get ERROR: .......................  NON-Manifold edges on models. Before I never got those. I went back and tried some older prints tat I did with no problem, now I get the error

### Expected results

Ability to print files

### Project file & Debug log uploads

[20-Painted Stock_1972_Chevrolet_Corvette_ZR1_1.zip](https://github.com/user-attachments/files/16829286/20-Painted.Stock_1972_Chevrolet_Corvette_ZR1_1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-09-03:
[image]
Hello, the non manifold error function has always been available. If you are a Windows user, you can right-click to find the repair function and fix it first.

Infinity-Papa-Joe-9mm, 2024-09-03:
I did try to right click and utilize the "FIX" option. The screen came up with the repair but would never complete the repair Thanks, Papa Joe Team Infinity

Haidiye00, 2024-10-08:
Hello, you can ignore it for now. We have checked your model and feel that the quality of the triangular facets in your model is not high. Could you please check where it was made and see if there is any room for improvement. We have checked that it does not affect printing.

[image]
and red lines is non manifold edges
[image]

closed 2024-10-16 by Haidiye00