bambulab/BambuStudio #4736: Unable to clear notification

jctronic, 2024-09-01:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.6.1

### Additional system information

Apple MacBook Pro 16"
CPU: Apple M1 Max
Ram: 32GB

### Printer

BambuLab X1C

### How to reproduce

1. Filament runs out

### Actual results

After filament spool replaced there is no way of clearing the notification on printer scree / mpbile app/ bambu studio

### Expected results

Option (button) to clear the notification on the printer screen and in mobile app / bambu studio

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-09-02:
[jctronic](https://github.com/jctronic)
[image]
In bambu studio，please click the icon to close the waring.This kind of pop-up window needs to be closed manually to disappear.

jctronic, 2024-09-02:
Yes that is not what Im talking about.
There is no way to clear the notification from here:

[image]
[image]

[image]