bambulab/BambuStudio #4737: Flow dynamic doesn't save & P1S restart

Tioborto, 2024-09-01:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonome 14.1.2

### Additional system information

16go Ram 
i7 

### Printer

Bambulab P1S 

### How to reproduce

1. Go to Calibration tab
2. Go to flow dynamics
3. Launch manual calibration 
4. Save it 

OR 

1. Go to Calibration tab
2. Go to flow dynamics
3. Create manual in history tab 
4. Save it 

### Actual results

Result doesn't appear in Calibration history

### Expected results

Calibration saved

### Project file & Debug log uploads

[debug_Fri_Aug_30_22_39_42_4934.log.0.zip](https://github.com/user-attachments/files/16829905/debug_Fri_Aug_30_22_39_42_4934.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Tioborto, 2024-09-05:
Hello @zhimin-zeng-bambulab , any news about this issue please ?

zhimin-zeng-bambulab, 2024-09-06:
It can be seen from the log that the last time you set the k value of "Elegoo - PLA - Black" was successful.
@Tioborto 
Could you reproduce this problem and attach the corresponding operation video, studio logs and printer logs? Thanks!

Tioborto, 2024-09-09:
Hello, Sorry I will upload all files. 
I try to save another k value of "another pla". 
[bambustudio_log.zip](https://github.com/user-attachments/files/16934221/bambustudio_log.zip)

Tioborto, 2024-09-09:
How to send to you log files and video ?

Tioborto, 2024-09-09:
Video : https://github.com/user-attachments/assets/fc66dabd-4bdd-48cc-a387-c9cebbeadb6e