bambulab/BambuStudio #4739: Crash on slicing with Intel Core I9

Grypus29, 2024-09-02:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Windows Version 10.0.22631 Build 22631
Processor :  Intel(R) Core(TM) i9-14900KF, 3200 MHz, 24 cœur(s), 32 processeur(s) logique(s)
Memory : DDR5 2x16Go 
MB : MAG Z790 TOMAHAWK MAX WIFI (MS-7E25)
SSD Samsung 980 Evo Pro

### Printer

Bambulab P1S

### How to reproduce

1 - Open a model with a high polygon amount
2 - Launch the slicing

### Actual results

Bambulab Studio Crashes.
The program close without any error message.
Sometimes, a popup appears before closing about memory access violation (Unable to get this popup again on the last tries)

### Expected results

Should calculate the slicing

### Project file & Debug log uploads

[Poly_Lotus_Sphere.zip](https://github.com/user-attachments/files/16835308/Poly_Lotus_Sphere.zip)
[crash_Mon_Sep_02_10_48_19_0_done.log](https://github.com/user-attachments/files/16835309/crash_Mon_Sep_02_10_48_19_0_done.log)
[crash_Thu_Aug_29_17_46_07_0_done.log](https://github.com/user-attachments/files/16835310/crash_Thu_Aug_29_17_46_07_0_done.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-09-02:
HI [Grypus29](https://github.com/Grypus29)：
Please try the following steps，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

QingZhangBambu, 2025-05-23:
close for no response

closed 2025-05-23 by QingZhangBambu