bambulab/BambuStudio #4741: overhang wall before the overhang bridge when slice

CanvasQAQ, 2024-09-02:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Additional system information

[image]
[image]

You can see that the hole location creates an overhanging wall before the bridge, which will directly gen - spaghetti .
The  model is https://makerworld.com.cn/zh/models/225710#profileId-150556
The gcode is as file
[gridfinity_miniwrench set bambu profile(3).gcode.zip](https://github.com/user-attachments/files/16839675/gridfinity_miniwrench.set.bambu.profile.3.gcode.zip)

### Printer

A1

### How to reproduce

As the gcode file/proj file, no need re-print, just review the slicer gcode you will find the issue.

### Actual results

spaghetti

### Expected results

normal print

### Project file & Debug log uploads

[gridfinity_miniwrench.zip](https://github.com/user-attachments/files/16839686/gridfinity_miniwrench.zip)

No log , just as the slicer screenshot

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-09-03:
@CanvasQAQ yep, it is an improvement point. but a bit trick to fix. I will add it to our feature list

QingZhangBambu, 2024-09-03:
@CanvasQAQ sorry, it needs support. the bridge and wall will fail no matter which is printed first.

MartinChan3, 2024-09-03:
@CanvasQAQ I think you should open the 'Use Support' option to solve this question...

CanvasQAQ, 2024-09-03:
Hi all, 

Thanks.
When I find that gen-spaghetti  then I add support and re-print, it have been print successed now.
I'm just curious that I have been prompted to open the option when this type overhanging wall appeared before (maybe 2weeks ago). But when sliced this model, I didn't get any warning. Of course, the main reason for this failure was that I didn’t check this model carefully.

Anyway, thank you very much.

BRs,
Canvas

closed 2024-10-12 by QingZhangBambu