bambulab/BambuStudio #4742: Issue with GCODE Generation in "Classic Wall Generation" Mode

Almisuifre, 2024-09-02:
### Bambu Studio Version

1.9.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processeur	13th Gen Intel(R) Core(TM) i9-13900KF, 3000 MHz, 24 cœur(s), 32 processeur(s) logique(s)

Mémoire physique (RAM) installée	64,0 Go
Mémoire physique totale	63,8 Go
Mémoire physique disponible	49,9 Go
Mémoire virtuelle totale	67,8 Go
Mémoire virtuelle disponible	45,6 Go
Espace pour le fichier d’échange	4,00 Go

Nom	NVIDIA GeForce RTX 4080
Mémoire vive sur la carte	(1 048 576) octets
Version du pilote	32.0.15.6094

### Printer

Bambu Lab X1C

### How to reproduce

Steps to reproduce the bug:
1. Open Bambu Lab Studio
2. Load the models
3. Slice and observe the issue (see image)

### Actual results

In "Classic Wall Generation" mode, I encountered an issue with GCODE generation. I have several identical objects, but one of them shows a kind of wall construction line in an area where there is no wall.

### Expected results

I repeated the same test using the "Arachnid Wall Generation" mode, and I did not observe the same issue.

I am submitting this issue in the hope that it can help in finding an adequate solution.

### Project file & Debug log uploads

[image]
[Rehausseur_X1C_V3_PETG_HF_Bug.zip](https://github.com/user-attachments/files/16840531/Rehausseur_X1C_V3_PETG_HF_Bug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-09-03:
@Almisuifre the clipper causes it, rotating the model a bit will fix that.

Almisuifre, 2024-09-03:
Ok, I understand. Yet it is the automatic placement that has turned it in this direction.

However, what is strange is that the arachnous mode is not affected.

Should we consider that the arachnene mode is now the norm?

QingZhangBambu, 2024-09-03:
Arachne mode uses different overhang detection methods.  some cases may fail on arachne, some only fail on classic

Almisuifre, 2024-09-03:
So, we choose according to the result to be obtained. I confess that I discover this new method of slicing.

Small question, probably a feature to do: Would it not be possible to trigger a warning when there is a positioning problem (rotation) as I had? This would prevent the distracted from creating a burgeoning impression.

QingZhangBambu, 2024-09-03:
we will consider that we are trying to improve.

Almisuifre, 2024-09-03:
Thank you very much to the whole team.

QingZhangBambu, 2024-09-03:
thanks for your support. you are so nice

QingZhangBambu, 2024-10-12:
@Almisuifre it has been fixed and will be released in the next version. please waiting

Almisuifre, 2024-10-14:
Good news.

Thanks

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz