bambulab/BambuStudio #4745: Bottom Surface running over Inner Wall

hossimo, 2024-09-03:
### Bambu Studio Version

1.09.05.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

i9 13900KF

### Printer

X1C

### How to reproduce

Download the [Filament Color Card With Tests](https://makerworld.com/en/models/599235#profileId-521344) and print layer 1.

The print does not suffer just looked odd

### Actual results

On layer 1 at about line 1312 the previous inner wall is run over by the bottom surface. Seems to print fine just not sure two layers should be printing ontop of eachother.

[image]

### Expected results

I would have thought it would either draw the bottom layer as solid or draw the bottom layer inside and outside of the inner wall

### Project file & Debug log uploads

[image]

[example-gcode.zip](https://github.com/user-attachments/files/16842006/example-gcode.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

maher-nakesh, 2024-09-03:
try this
https://bit.ly/4e4zLlq
select the desired compiler in the installer

QingZhangBambu, 2024-09-03:
@XunZhangBambu please check this, the bottom area is not right.

XunZhangBambu, 2024-09-11:
@hossimo  Disable the “Only one wall on top surface”
[image]

hossimo, 2024-09-14:
"Only one wall on first layer" or "Only one wall on top surfaces" = Not applied both work. Thanks!