bambulab/BambuStudio #4746: Vertical stripes on fuzzy skin in Bambu studio ver1.09.05.51

supuri, 2024-09-03:
### Bambu Studio Version

1.09.05.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows10

### Printer

P1S and A1 mini

### How to reproduce

Fuzzy skin>Contour.
Maybe just selecting a fuzzy skin will happen with any of them.

It seems to me that this symptom is more likely to appear in geometries that use "sweeps" on CAD.

### Actual results

[image]
[image]

Vertical stripes are already present in the preview stage of Bambu Studio.

### Expected results

[image]
[image]

Ver 1.09.03.50 did not include vertical stripes. I would like to see it return to this state.

### Project file & Debug log uploads

Just screenshots and pics.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-09-03:
HI  [supuri](https://github.com/supuri)：
Could you please provide the source file,we will check it.

supuri, 2024-09-03:
[Test1.gcode.zip](https://github.com/user-attachments/files/16843397/Test1.gcode.zip)

Here it is, it sliced in Ver 1.09.05.51. Stripes appear.

vahurvilta, 2024-09-13:
I have exact same problem after updating.

huangdun, 2024-09-15:
Also seeing this strange line on my print when fuzzy skin is used.

ttt888zzz, 2024-09-18:
Thank you for your feedback. We will update this feature in the next release soon

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz